ROBOTIS-GIT/turtlebot3 #164: TurtleBot Bringup Not Connecting to Port

nprof50, 2018-03-30:
I've been using the TurtleBot3 Waffle with Intel Joule processor for about a month now, and everything has been working very smoothly up until yesterday. Yesterday I followed the same process I usually use to set the TurtleBot up to communicate with a Remote PC:

[Remote PC]

1. Run `roscore`

2. Run `roslaunch turtlebot3_bringup turtlebot3_remote.launch`

[TurtleBot]

1. Power up the robot

2. Run `roslaunch turtlebot3_bringup turtlebot3_robot.launch`

After I run these steps, I usually am able to publish/subscribe to the various ROS topics the robot publishes/subscribes to. Yesterday I ran this setup and received this console output after launching the turtlebot3_robot.launch on the TurtleBot:

    Checking log directory for disk usage. This may take awhile.
    Press Ctrl-C to interrupt
    Done checking log file disk usage. Usage is <1GB.
    
    started roslaunch server http://[hostname]:45747/
    
    SUMMARY
    ========
    
    PARAMETERS
     * /rosdistro: kinetic
     * /rosversion: 1.12.13
     * /turtlebot3_core/baud: 115200
     * /turtlebot3_core/port: /dev/ttyACM0
     * /turtlebot3_lds/frame_id: base_scan
     * /turtlebot3_lds/port: /dev/ttyUSB0
    
    NODES
      /
        turtlebot3_core (rosserial_python/serial_node.py)
        turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
        turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)
    
    ROS_MASTER_URI=http://[master_uri]:11311
    
    process[turtlebot3_core-1]: started with pid [2603]
    process[turtlebot3_lds-2]: started with pid [2604]
    process[turtlebot3_diagnostics-3]: started with pid [2605]
    [INFO] [1522418950.044044, 0.000000]: ROS Serial Python Node
    [INFO] [1522418950.263477, 0.000000]: Connecting to /dev/ttyACM0 at 115200 baud

<br/>
And that's as far as it gets. It's getting hung up on connecting the OpenCR port. The only similar scenario to mine that I've seen asked about on here is at the following link: (https://github.com/ROBOTIS-GIT/turtlebot3/issues/69). I've followed all the instructions in the answers provided with no successful.

Strangely enough, I let the robot charge overnight, and when I powered it on again this morning, it worked fine for the first 3 or 4 times I ran the boot sequence above. Then I tried to start again after 3-4 successes, and I ran into the same issue. It gets stuck on the connecting step.

I am able to connect to the OpenCR board and flash the turtlebot3_core firmware from the Arduino program, so I know I have connectivity to the board. It's just the turtlebot3_robot.launch sequence that seems to be having trouble with connecting. I'm planning to re-flash the BIOS for the Intel Joule processor being used alongside the OpenCR board. I have tried powering on and off, unplugging and plugging back in every port possible to try and diagnose the issue, but I haven't had any success.

If more information is needed, I would be happy to provide it. Thanks for the help.

routiful, 2018-04-02:
Hi :)

Which version of firmware do you used? The latest version of it is 1.1.2.

nprof50, 2018-04-02:
If you're referring to the firmware I flashed to the OpenCR board, I'm pretty sure it's the latest. I flashed it through the Arduino program as instructed in the manual, and the tools I used are all up-to-date. Is there a simple way for me to confirm that?

routiful, 2018-04-03:
Did you try update and upgrade command?

```
 $ sudo apt-get update
 
 $ sudo apt-get upgrade
```

nprof50, 2018-04-10:
Yes, I've run an update/upgrade. I replaced the Intel Joule board with a Raspberry Pi, and the turtlebot was working fine for about a week. Today it started having the exact same issue as the Intel Joule board, so I've concluded something's wrong with the OpenCR board. The problem remains that I can flash firmware to the OpenCR board from the Pi or Joule (even lets me run a test "blink" program to ensure it works), but I cannot get the turtlebot_robot.launch sequence to complete past the "Connecting to /dev/ttyACM0" step.

routiful, 2018-04-11:
I'm sorry for inconvenience. Please contact customer support team in ROBOTIS. Then they inform you how to exchange it . contactus2@robotis.com

closed 2018-04-16 by routiful