ROBOTIS-GIT/turtlebot3 #165: no transform form of the robot model after running Rviz

jefferycaojin, 2018-04-03:
Hello :)
 we have followed the emanual to setup the turtlebot3 Burger and the environment of Rviz with no error. After running Rviz, we encountered the problem that there is no transform form of the model to the map in rviz.
We tried the solutions of similar issue on [https://github.com/ROBOTIS-GIT/turtlebot3/issues/137](url), while the error exist anyway.  how to find where the problem is ?
here is the error message in terminal :
[image]

and error message in rviz window:
[image]
[image]

THANKS.
Jeffery

routiful, 2018-04-03:
Which launch file do you used?

**turtlebot3_remote** or **turtlebot3_model** ??

jefferycaojin, 2018-04-04:
@routiful 
Thanks for your reply!
Here is our command.
[Remote PC]
$ roscore
$ export TURTLEBOT3_MODEL=burger
$ roslaunch turtlebot3_bringup turtlebot3_remote.launch
$ rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz
[TurtleBot] 
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch

We have followed the instructions on [http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#pc-setup](url).
Development environment is as below:
pc : ubuntu 16.04
ROS : ROS kinetic Kame
Raspberry Pi 3: ubuntu mate 16.04 

Thanks,
Jeffery

jefferycaojin, 2018-04-06:
@routiful 
Grateful for your reply.
we will check if the graphic driver is working well since we establish the ubuntu 16.04 on the macbook air (macOS). Maybe the graphic driver is not compatible with the linux environment.
We would try another pc to see if that's the fatal problem.

Thanks,
Jeffery

jefferycaojin, 2018-04-09:
We encountered the same error on another pc(win10) which had the graphic driver installed.
While the problem is settled after setting up the Fixed Frame under Global Options of the Displays, which is in the red rectangular area showed in the picture below.

[image]

Thank you for your kind help!
Jeffery

closed 2018-04-09 by jefferycaojin

routiful, 2018-04-10:
Great. Has the warning messages gone after installed driver?

jefferycaojin, 2018-04-11:
Hi, thanks for your concern. The warning messages in Rviz are gone after we set up the Fixed Frame which we ignored before. It seems that the warning messages about the STL files don't cause any problem.
Thanks,
Jeffery

florianspy, 2020-05-19:
i experienced a similar problem with Transform For frame [map]: No transform to fixed frame [base_link]. TF error: [Lookup would require extrapolation into the past. Requested time 1455208633.090277046 but the earliest data is at time 1589896389.641628849, when looking up transform from frame [map] to frame [base_link]]
for me the reason was that my turtlebot and my PC had different timesettings so i had to sync them again and the error was gone.

naseef139, 2020-05-21:
[ERROR] [1590092344.403659387]: link 'base_link' is not unique.
and Rviz shows status error in Robot model
can anyone help me to solve this????

shmpwk, 2020-06-16:
@naseef139 Maybe there are two or more 'base_link'. Check your robot model file s.t. urdf or sdf file.