ROBOTIS-GIT/turtlebot3 #168: Differential drive control

zweistein, 2018-04-06:
Is there an easy way to use the robot driver on the turtulebot3 burger such that the motion control is differential drive rather than operation space control (ROS cmd_vel)?

In order to control the wheel velocity directly, do I need to use the dynamixel drivers from  dynamixel_workbench controller?

routiful, 2018-04-08:
Hi :)

Dynamixel in TB3 is controlled by [Dynamixel SDK](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/include/turtlebot3/turtlebot3_motor_driver.h) that is included in turtlebot3_core.ino (firmware).  

You can make other subscriber to control it with [writeVelocity()](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/include/turtlebot3/turtlebot3_motor_driver.h#L70) function

zweistein, 2018-04-09:
Thanks for the advice, I would like to use a service with a custom message type (or similar topic) like:
```
float32 left
float32 right
---
bool success
``` 

Can I also create a service server in the `tb3.core`? 
Can I use custom `.srv` or `.msg` in the core?
Are there any examples regarding this feature extension?

zweistein, 2018-04-10:
Thanks for the help, I managed to build in the Arduino IDE a new firmware that includes my message type. Before downloading this new firmware to the board, is there a way to upload form the board to the PC the current firmware to keep it as a backup?

routiful, 2018-04-11:
I'm sorry,unfortunately, there is no way to backup it after downloaded.

zweistein, 2018-04-16:
I uploaded the new firmware with `my_DiffDrive` massage. But somehow I have made an arduino  `ros_lib` that does not match the one that is on the **RPi** because  when I launch the driver `turtlebot3_robot.launch` I get an error: 

This seems to be located in the: `SensorState` msg 
The new firmware of the opencr was built on a **PC** and not on the **RPi** but the `turtulebot3_msgs` pkg has the same version number on bot machines (v 0.1.5) 

How can I fix this issue? Where is the mismatch between the two massages?

zweistein, 2018-04-16:
process[master]: started with pid [5291]
ROS_MASTER_URI=http://192.168.0.1:11311/

setting /run_id to a63083e0-d0df-11e5-9b67-b827ebcfff6c
WARNING: Package name "melodyPlayer" does not follow the naming conventions. It should start witha lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[rosout-1]: started with pid [5304]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [5315]
process[turtlebot3_lds-3]: started with pid [5322]
process[turtlebot3_diagnostics-4]: started with pid [5323]
[INFO] [1455209472.125106]: ROS Serial Python Node
[INFO] [1455209472.147911]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1455209474.303776]: Creation of publisher failed: Checksum does not match: d537ed7b8d95065b6c83830430b93911,427f77f85da38bc1aa3f65ffb673c94c

routiful, 2018-04-17:
Hi :)

If you want to add your own msg in Arduino. You have to make new `ros_lib`

First, make `msgs` package in your remote **PC** like [turtlebot3_msgs](https://github.com/ROBOTIS-GIT/turtlebot3_msgs)
You should concern `CMakeLists` to build properly.

Second,  make new `ros_lib`. Below link would be help you http://wiki.ros.org/rosserial_arduino/Tutorials/Arduino%20IDE%20Setup

Third, replace `[turtlebot_ros_lib](https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/turtlebot3_ros_lib)` to all files you made before

Fourth, build it by Arduino IDE and upload to OpenCR

Fifth, copy and paste `msgs` package in your **RPI** and cmake it.

Good luck!

zweistein, 2018-04-20:
Thank you for your help. I managed to fix the message issue by following your instructions and regenerating the ROS_lib. 

Now, the new driver is working flawlessly :)

closed 2018-04-20 by zweistein