ROBOTIS-GIT/turtlebot3 #173: OpenCR does not boot

yoann-solana, 2018-04-10:
Hello,

First of all thank you for the work you have done and you are doing, Turtlebot3 is a great project.

I am working on the Turtlebot3 waffle pi. I followed all the steps from the emanual to setup the hardware but at this [step](http://emanual.robotis.com/docs/en/platform/turtlebot3/hardware_setup/#basic-operation) the robot is not moving when the SW1 and SW2 push buttons are pressed.

I have checked every connections several times.
I can see on the board only the USB and POWER leds are on).

However, everything else seems to work properly (lidar, BT-410, raspberry pi (ssh connection working)).
I tried to detect the connection between the raspberry pi and the OpenCR over USB:
- with the IDE arduino but there was no ACM0 port displayed
- with lsusb command but nothing appeared related to the board 

I also tried with a personal USB cable between the OpenCR and my laptop, but the result was the same.

In addition, I tried to use the [recovery mode](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#firmware-recovery-mode) but there was no activity from the board (the STATUS led remains off).

Do you have any idea to solve this problem?

Thanks in advance,
Yoann

routiful, 2018-04-11:
Hi :)

Please check whether OpenCR has proper bootloader. 
You need to [Run DFU Mode](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#run-dfu-mode) and download bootloader to it. After that you can find ACM0.

yoann-solana, 2018-04-11:
Hi,
Thanks for the fast answer.
I tried to enter in DFU mode but nothing append. The lsusb command does not show the STM device

$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 5986:066d Acer, Inc 
Bus 001 Device 004: ID 0bda:b002 Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 1c7a:0603 LighTuning Technology Inc. 
Bus 001 Device 002: ID 048d:1170 Integrated Technology Express, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Do you have another idea?

routiful, 2018-04-12:
Hi again :)

Sorry inconvenience. I assume that is caused by hardware fault.
You can contact to customer support team in ROBOTIS (contactus2@robotis.com) to exchange it.

Before contact them, please check again your USB cable and follow [Wiki](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#arduino-ide) once again 

Thanks

yoann-solana, 2018-04-12:
I have checked with other USB cables and followed the Wiki again but it is the same result.
I will contact the customer support.

Anyway thanks for the help.

Regards.

closed 2018-04-12 by routiful

timdori, 2018-08-17:
Did you get a solution for your problem?