ROBOTIS-GIT/turtlebot3 #178: Parameters for narrow pathway

arixrobotics, 2018-04-13:
Hi, 
I'm trying to tune my navigation parameters so that the turtlebot3 can go though pathways of ~30cm. Its a maze, with lots of turns and corners. 
As seen in this video, it keeps oscillating back and forth at such corners. 
Do you have any idea on how to solve this?
[https://youtu.be/M-YFEJumX5s](url)
[image]

Arif

routiful, 2018-04-16:
Hi :)

You would check inflation radius in [costmap_common_params_burger.yaml ](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_navigation/param/costmap_common_params_burger.yaml#L5)

It makes more sharper layer around obstacle near TB3. If you want to get more information about it please follow the link below

http://wiki.ros.org/costmap_2d

arixrobotics, 2018-04-16:
Thanks for the reply @routiful . But I think its more than just making the inflation 'slimmer'. As in the previous video, I've made the inflation to be only about 2cm, but the planner still had a hard time. 

So I've been playing around with the dwa parameters, and found that these made it work:
  path_distance_bias: 0.01 # default:32.0
  goal_distance_bias: 0.05 # default:24.0
  occdist_scale: 0.01 # default:0.04

See video here: https://www.youtube.com/watch?v=AamHifhvNMs

closed 2018-04-16 by arixrobotics

robotpilot, 2018-04-16:
I saw your video. I remembered memories of a previously popular maze of MicroMouse.
I am also very interest how your project will be used in the future. :)
Thank you for your work and I hope to watch your work.
https://youtu.be/AamHifhvNMs