ROBOTIS-GIT/turtlebot3 #187: /dev/tty/ACM0 disappears when OpenCR's 12V output is used

evbernardes, 2018-04-19:
I'm trying to connect a custom circuit board to the OpenCR card, using the 12V output show at the left of the following image as a source (this custom board doesn't take up more than some miliamps of current).

[image]

Nevertheless, at the moment that I connect to this output, the USB connection from the OpenCR board to the Raspberry Pi is lost. Using `$ watch ls /dev/ -m` I was able to see that it indeed happens at the very moment I plug it in.

Plugging and unplugging the USB cable does nothing, the device can only be seen after turning the OpenCR board off and on again.

What could be the issue? Thanks in advance for any tips that might help me to identify the problem.

ROBOTIS-Will, 2018-04-24:
@evbernardes 
First of all, check your power supply is providing enough power. I sometimes get a bad adapter that does not provide enough current written on it.
If you are already supplying power with decent power supply or battery, please check whether the 3 analog input signals from your card do not exceed 5V.
Although 12V and 5V port are shared power from a 12V regulator, they should be able to supply enough power to RPI3 and your custom card that only consumes several miliamps.

evbernardes, 2018-04-24:
Hello, @ROBOTIS-Will!
The power supply is indeed ok (but since it didn't seem to be the problem, I'm trying to do my tests using the power from the "Power Out" output).

The analog input signals don't exceed 5V. If they did, though, how could that be a problem for the USB port?

Thanks a lot for your attention!

ROBOTIS-Will, 2018-04-25:
@evbernardes 
Thanks for checking the system.
Well, I thought if the analog input signal exceeds 5V, then it might reset or enable hardware protection from MCU to lead such issue.
Have you tried reading your custom card on other boards?

evbernardes, 2018-04-25:
@ROBOTIS-Will 
I see, can you point me some document regarding the MCU in which I can try to study its behavior? 

I have used the same custom card with an Arduino Uno board and I didn't have this problem. Moreover, I have managed to use it with the Turtlebot3: sometimes it works for hours with no problem and then it just randomly stops, sometimes it just doesn't work since the beginning.

Thanks again for you help!

closed 2018-05-02 by robotpilot