ROBOTIS-GIT/turtlebot3 #192: Waffle Plate 3D-CAD data on OnShape

roboshiba, 2018-04-24:
TB3 3D-CAD datas that linked from e-manual are partially different from actual product.
Please update formal data of Waffle Plate.
It is necessary for optional camera mount and extended parts design by user own.
Thanks.

[image]

[image]

routiful, 2018-04-25:
Hi :)

3D printable data of Waffle plate is different to Off-the-shelf data of it. Because Off-the-shelf data isn't compatible with 3D printing. But data you commented has identical specification with formal data.
If someone wanted to design for extended parts, they could believe uploaded data on Onshape.

Thanks

roboshiba, 2018-05-01:
I understood!
That's excellent reason.
If you can, PLS add this memo to e-manual.

Thanks.

closed 2018-05-01 by roboshiba