ROBOTIS-GIT/turtlebot3 #193: Opencr board not connecting through usb and overheating

hdstusnick, 2018-04-24:
My name is Hunter and I am working on a group project and my teammate Michael has called recently. We are working on an opencr board with a pololu shield. Today when I sat down to work on the opencr board, it would not connect to a desktop or the pi that normally runs it. It also gets very hot quickly when plugged into power or the computer via usb. I have run into the board not connecting to the computer before but have always fixed it by holding the sw2 button and then pressing the reset button. Now that does not work. Trying to get it into DFU mode has also been unsuccessful. Any advice on how to salvage the board so I upload code to it again would be helpful.

ROBOTIS-Will, 2018-04-25:
Hello Hunter @hdstusnick,
When there's a short circuit, the board usually gets very hot in a specific IC, and in your case, it could be the MCU as it does not allow to boot in DFU mode.  
I'm not 100% sure the whether the MCU is damaged or not, but I recommend to [send it for a repair](http://en.robotis.com/model/page.php?co_id=repaircenter) to see if it can be repaired.  
Thank you.

hdstusnick, 2018-04-25:
Hi Will,

Thank you for getting back to me quickly and for your help in this matter.
I am a student working on a robotics project at Brandeis University and am
getting very close to a deadline. I am going to send the board in for
repair and I just wanted to know if there was any way to expedite the
repair process or if we could have a loaner in the meantime.

Best,
Hunter

ROBOTIS-Will, 2018-04-26:
Hi @hdstusnick ,
Repairing the board could take a couple weeks as it has to make an oversea trip.  
Please contact at cs@robotis.com to see if they can help you.  
Thank you.

closed 2018-04-29 by robotpilot

luqmansen, 2020-09-07:
> When there's a short circuit, the board usually gets very hot in a specific IC, and in your case, it could be the MCU as it does not allow to boot in DFU mode.
> I'm not 100% sure the whether the MCU is damaged or not, but I recommend to [send it for a repair]

Hi, seems I encountered the exact same problem (MCU super hot) except, mine is still detected as OpenCR board on the computer. What bugging me the most was, if there is a short circuit (moreover in the MCU), how it is still able to be detected and work normally (except it is super hot and I have to turn it off after a few seconds since I'm afraid it'll be burned).

Thank you in advance

ROBOTIS-Will, 2020-09-08:
@luqmansen 
Hi, running high resource required firmware can also heat up the MCU.
Leave the board turned on with the DFU mode and see if the MCU gets hot.
Depending on short circuit conditions, MCU may or may not work.
I'd recommend to first check if there's any dust or conductive debris on the board.
Thank you.

luqmansen, 2020-09-08:
Hi @ROBOTIS-Will, thanks for your quick reply. I confirmed that I'm not running a high resource program since I'm only running a simple example from the Robotis repository.

Also I cannot get into DFU mode. The MCU just literally gets super hot as soon as I plugged in to the USB port.

I never experienced this "partial" broken MCU like this. Usually if it's gets shorted then it just died. Yes we'll make sure there is no conductive debris on the board.

ROBOTIS-Will, 2020-09-09:
@luqmansen 
You can also request the RMA service for repairing the board.
If it is still under warranty, I recommend to request the service so that our engineers can investigate the board.
Thank you.