ROBOTIS-GIT/turtlebot3 #194: Odom doesn't move.

PM201472, 2018-04-26:
The robot's odom wont move. When using the teleop, everything moves except the odom.

routiful, 2018-04-27:
Hi :)

Would you tell me more information about your environment? such as software and firmware version and which launch file you execute before running it and terminal window when you faced error state.

PM201472, 2018-04-30:
Firmwear: 1.1.2
ROS: Kinetic

Steps to run:
<pc>
1. roscore
<bot>
2. roslaunch turtlebot3_bringup turtlebot3_lidar.launch
3. roslaunch turtlebot3_bringup turtlebot3_core.launch
<pc>
4. roslaunch turtlebot3_bringup turtlebot3_remote.launch
5. rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz

routiful, 2018-05-02:
Have any error or warning message in your terminal?

PM201472, 2018-05-02:
[ INFO] [1525280086.822428808]: rviz version 1.12.15
[ INFO] [1525280086.822474352]: compiled against Qt version 5.5.1
[ INFO] [1525280086.822483668]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1525280087.050782949]: Stereo is NOT SUPPORTED
[ INFO] [1525280087.050858555]: OpenGl version: 3 (GLSL 1.3).
[ WARN] [1525280087.121801169]: The STL file 'package://turtlebot3_description/meshes/bases/burger_base.stl' is malformed. It starts with the word 'solid', indicating that it's an ASCII STL file, but it does not contain the word 'endsolid' so it is either a malformed ASCII STL file or it is actually a binary STL file. Trying to interpret it as a binary STL file instead.
[ WARN] [1525280088.840902900]: The STL file 'package://turtlebot3_description/meshes/sensors/lds.stl' is malformed. It starts with the word 'solid', indicating that it's an ASCII STL file, but it does not contain the word 'endsolid' so it is either a malformed ASCII STL file or it is actually a binary STL file. Trying to interpret it as a binary STL file instead.

routiful, 2018-05-03:
This message will not help me. Please show me turtlebot3_core terminal.

PM201472, 2018-05-03:
... logging to /home/pi/.ros/log/b0492804-4ee7-11e8-9bbd-1c1b0d9b1f73/roslaunch-raspberrypi-1599.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.19.201.146:34967/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)

ROS_MASTER_URI=http://172.19.142.166:11311

process[turtlebot3_core-1]: started with pid [1621]
[INFO] [1525285373.729015]: ROS Serial Python Node
[INFO] [1525285373.783443]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1525285376.092533]: Note: publish buffer size is 1024 bytes
[INFO] [1525285376.094407]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1525285376.106396]: Setup publisher on version_info [turtlebot3_msgs/VersionInfo]
[INFO] [1525285376.126255]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1525285376.142893]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1525285376.209115]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1525285376.221837]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1525285376.233295]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1525285376.245731]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1525285380.941992]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1525285380.961517]: Note: subscribe buffer size is 1024 bytes
[INFO] [1525285380.963210]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1525285380.986353]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1525285381.015247]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1525285381.034383]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1525285381.042790]: --------------------------
[INFO] [1525285381.047039]: Connected to OpenCR board!
[INFO] [1525285381.050795]: This core(v1.1.2) is compatible with TB3 Burger
[INFO] [1525285381.054396]: --------------------------
[INFO] [1525285381.058012]: Start Calibration of Gyro
[INFO] [1525285381.546246]: Calibration End

yiorgosk, 2018-05-04:
I am having the same issue, i can't move the turtlebot even though i did the turtlebot3_motor_setup and uploaded the turtlebot3_core. My firmware is also 1.1.2 .

routiful, 2018-05-08:
@PM201472 @yiorgosk 

Hi, sorry for late reply.

Check /odom topic below message to show how it works

`$ rostopic echo /odom`

And open rviz and teleop

```
$ rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz
$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
```

Change **Fixed Frame** to `odom` in **Global Options**

Then you may show your turtlebot3 is running in Rviz.

Thanks.

robotpilot, 2018-05-23:
Has this issue been resolved?

PM201472, 2018-05-25:
No it has not.

routiful, 2018-05-28:
Sorry for late reply

odom is calculated in [turtlebot3_core.ino](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino#L428) and publish it.

Would you try to upload new firmware (v1.2.0) and software (v1.0.0)?
Below link might be help you

https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888/2

Thanks

closed 2018-06-03 by robotpilot