ROBOTIS-GIT/turtlebot3 #197: TB3 Waffle(not Pi) bring up error

roboshiba, 2018-05-02:
User tried Waffle's bring up in order of tutorial, but got an error and didnt move on.

-Tutorial Step 9 has been completed. Error occurs in [ roslaunch turtlebot 3 _ bringup turtlebot 3 _ robot.launch] of tutorial 9.1.2.
http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup

-The model of bashrc is "waffle".

-Firmware has been updated.

Is there any way to solve it?

-----error info
[INFO] [1525081305.385632]: Setup publisher on joint_states [sensor_msgs/JointState]

[INFO] [1525081305.388253]: Setup publisher on battery_state [sensor_msgs/BatteryState]

[INFO] [1525081305.390948]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]

[INFO] [1525081305.393436]: Setup publisher on /tf [tf/tfMessage]

[ERROR] [1525081305.408461]: Tried to publish before configured, topic id 129

[ERROR] [1525081305.422598]: Tried to publish before configured, topic id 129

[INFO] [1525081305.437400]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]

[INFO] [1525081305.439970]: Setup publisher on version_info [turtlebot3_msgs/VersionInfo]

[INFO] [1525081305.443438]: Setup publisher on imu [sensor_msgs/Imu]

[INFO] [1525081305.446295]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]

[ERROR] [1525081305.449654]: Creation of publisher failed: nav_msgs

ROS path [0]=/opt/ros/kinetic/share/ros

ROS path [1]=/home/ais/catkin_ws/src

ROS path [2]=/opt/ros/kinetic/share

[INFO] [1525081305.452323]: Setup publisher on joint_states [sensor_msgs/JointState]

[INFO] [1525081305.454889]: Setup publisher on battery_state [sensor_msgs/BatteryState]

[INFO] [1525081305.457870]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]

[INFO] [1525081305.460536]: Setup publisher on /tf [tf/tfMessage]

routiful, 2018-05-03:
Hi Shibata san:)

This error was caused by mismatch version between firmware and software.
Please check firmware(v1.1.2) and software(master branch)

Thanks

robotpilot, 2018-05-09:
@roboshiba, please check the answer of @routiful.
If the answer is solved, please close this issue. And if it is not enough, please comment. :)

robotpilot, 2018-05-11:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-05-11 by robotpilot

roboshiba, 2018-05-23:
Sorry, I heard solved it on last week.

Question about TB3, especially from students.
We will recommend as written directly by them in Issue.
This thread was made Example.

Thanks