ROBOTIS-GIT/turtlebot3 #198: Navigation launch file

PujieXin, 2018-05-04:
Hi,

When I use navigation part, I need to roslaunch the turtlebot3_navigation.launch. When I check the launch file, I can see pkg="map_server", pkg="move_base" etc. 

My question is that, I didn't find these pkg code( cpp and python) in Turtlebot3, are they given by ROS directly? Because when I learnt navigation part, the navigation launch file are totally the same.

Thank you!

kijongGil, 2018-05-08:
Hi :)
Yes, these pkg are standard. If you want see these code, see link below.

move_server : http://wiki.ros.org/map_server

move_base : http://wiki.ros.org/move_base

routiful, 2018-05-08:
Hi :)

turtlebot3_navigation launch file includes `acml` xml filed and calls `move_base` node on `move_base` package with own navigation parameters.

As you mentioned, the launch file looks same with other file that is provided ROS. However, the parameters are totally different as appearance of robot or laboratory environments.

You can find base code below link

https://github.com/ros-planning/navigation

Thanks

closed 2018-05-14 by robotpilot