ROBOTIS-GIT/turtlebot3 #199: Can't use rviz turtlebot3 slam configuration file

maximilienpe, 2018-05-07:
Hi ! 

I can't use the configuration file in turtlebot3_slam. It gives me a segmentation fault with Rviz. 

OS of the PC : Debian Jessie
OS of the raspberry pi 3 b : Ubuntu Xenial

Process : 
(1) PC : roscore
(2) Turtlebot3 : roslaunch turtlebot3_bringup turtlebot3_robot.launch

================================================================================
... logging to /home/turtlebot3/.ros/log/e14a5510-51e7-11e8-887e-b8ac6f43b528/roslaunch-turtlebot3-desktop-2660.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.128.128:42388/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.128.68:11311

process[turtlebot3_core-1]: started with pid [2688]
process[turtlebot3_lds-2]: started with pid [2689]
process[turtlebot3_diagnostics-3]: started with pid [2690]
[INFO] [1525695852.748995]: ROS Serial Python Node
[INFO] [1525695852.813717]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1525695854.977798]: Note: publish buffer size is 1024 bytes
[INFO] [1525695854.979438]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1525695854.989607]: Setup publisher on version_info [turtlebot3_msgs/VersionInfo]
[INFO] [1525695855.000042]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1525695855.012410]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1525695855.039606]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1525695855.052477]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1525695855.064908]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1525695855.074904]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1525695857.354180]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1525695857.379500]: Note: subscribe buffer size is 1024 bytes
[INFO] [1525695857.380958]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1525695857.405518]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1525695857.431306]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1525695857.446810]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1525695858.012520]: --------------------------
[INFO] [1525695858.015653]: Connected to OpenCR board!
[INFO] [1525695858.018791]: This core(v1.1.2) is compatible with TB3 Burger
[INFO] [1525695858.021902]: --------------------------
[INFO] [1525695858.024968]: Start Calibration of Gyro
[INFO] [1525695860.568680]: Calibration End

================================================================================

(3) PC : roslaunch turtlebot3_slam turtlebot3_slam.launch

================================================================================
[... 341 lines cut ...]
[ INFO] [1525697874.134825133]: Parsing script point_cloud_tile.material
[ INFO] [1525697874.135172155]: Parsing script point_cloud_point.material
[ INFO] [1525697874.135493083]: Parsing script default_pick_and_depth.material
[ INFO] [1525697874.135874451]: Parsing script point_cloud_square.material
[ INFO] [1525697874.136156561]: Parsing script point_cloud_sphere.material
[ INFO] [1525697874.136447321]: Parsing script point_cloud_box.material
[ INFO] [1525697874.136705509]: Parsing script point_cloud_flat_square.material
[ INFO] [1525697874.136986055]: Parsing script liberation_sans.fontdef
[ INFO] [1525697874.137111343]: Finished parsing scripts for resource group rviz
[ INFO] [1525697874.137152555]: Creating resources for group rviz
[ INFO] [1525697874.137187954]: All done
[ INFO] [1525697874.139920365]: GLRenderSystem::_createRenderWindow "OgreWindow(1)", 100x30 windowed  miscParams: FSAA=4 contentScalingFactor=1,000000 externalGLControl= externalWindowHandle=50331673 parentWindowHandle=50331673 
[ INFO] [1525697874.194255808]: GLXWindow::create used FBConfigID = 277
[ INFO] [1525697874.242327214]: Mesh: Loading rviz_sphere.mesh.
[ INFO] [1525697874.261355245]: Texture: SelectionRect0Texture: Loading 1 faces(PF_R8G8B8A8,1x1x1) Internal format is PF_A8R8G8B8,1x1x1.
[ INFO] [1525697874.264591482]: Mesh: Loading rviz_cylinder.mesh.
[ INFO] [1525697874.264949229]: Mesh: Loading rviz_cone.mesh.
[ INFO] [1525697874.283044890]: Texture: ROSImageTexture0: Loading 1 faces(PF_A8R8G8B8,420x300x1) Internal format is PF_A8R8G8B8,420x300x1.
[ INFO] [1525697874.283556842]: Texture: MapPaletteTexture0: Loading 1 faces(PF_A8B8G8R8,256x1x1) Internal format is PF_A8R8G8B8,256x1x1.
[ INFO] [1525697874.283700325]: Texture: MapPaletteTexture1: Loading 1 faces(PF_A8B8G8R8,256x1x1) Internal format is PF_A8R8G8B8,256x1x1.
[ INFO] [1525697874.283787339]: Texture: MapPaletteTexture2: Loading 1 faces(PF_A8B8G8R8,256x1x1) Internal format is PF_A8R8G8B8,256x1x1.
Erreur de segmentation

================================================================================
Maximilien

routiful, 2018-05-08:
Hi :)

You can fix malformed stl after you launch fourth commands

https://gist.github.com/awesomebytes/2595b1dc41831c804a4f

We will solve that problem next update. Sorry for inconvenience.

Thanks.

maximilienpe, 2018-06-07:
Hi, I had no answer on ROS Answer about my question and the trouble. 

Today, I have installed ROS Kinetic on another computer on Debian 8 Jessie, 
with a better graphic card and so a newer version of OpenGL and I have the same results. 
When I use Rviz with your configuration file it still crashing with a segmentation fault. 

Another information I have for you, since the last update of May the 25th, it crashes not 
only for the configuration file of the SLAM part but also with the other in description, 
and in navigation). 

Have change anything in those configuration file which can give us a clue about what 
is the trouble with Debian 8 ? 

Thank you in advance for all your answers.