ROBOTIS-GIT/turtlebot3 #201: Replacing raspberry PI with Upboard

florianspy, 2018-05-11:
ISSUE TEMPLATE ver. 0.0.1

1. Which TurtleBot3 you have?

- [ ] burger
- [ ] waffle
- [x ] waffle pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

- [ ] Raspberry Pi 3
- [ ] Intel Joule
- [ x] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

- [ ] Ubuntu MATE 16.04 or later
- [ ] Raspbian
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in RemotePC?

- [ x] Ubuntu 16.04 LTS (Xenial Xerus)
- [ ] Ubuntu 18.04 LTS (Bionic Beaver)
- [ ] Linux Mint 18.X
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

I am planning to use an Upboard Squard in Combination with TurtleBot 3 waffle PI.
So my question is if i can simply use the installation Guide provided for the intel joule to bring it up and working.
The main issue with the raspberry PI is its limitation on usb 2.0.

routiful, 2018-05-14:
Hi :)

I don't understand your main issue. Do you want to replace Raspberry Pi 3 to Upboard? What limitation on usb 2.0 on Raspberry Pi 3??

Thanks

florianspy, 2018-05-14:
My main goal is as you said to replace the raspberry pi 3 with an Up Board Squared.
The reason for me is the lack of the raspberry pi of USB 3.0 ports. Thats why i stated limitation on usb 2.0. Sorry for the confusion.
Also i would like to have more computing power than the raspberry can offer.
Thank you for your help.

robotpilot, 2018-05-17:
We also tested version of TurtleBot3 equipped with Up Board. If you only care about the power supply to Up Board, you can use the same software without any problems.

robotpilot, 2018-05-20:
Yes. Up Board is the same in software. There is no problem. You only have to handle the way of supplying power to the Up Board by hardware.

florianspy, 2018-05-22:
As you already tested the TurtleBot 3 with the Up Board may i ask for any recommendation regarding the power supply?

robotpilot, 2018-05-22:
@florianspy 
We provide Raspberry Pi 3 Power Cable. This cable supplies OpenCR's 5V power to Raspberry Pi 3.
[image]
If you use an Upboard, you must make and use a power cable for the SBC. The OpenCR provoide 3.3V, 5V, and 12V in TurtleBot3. In the case of Upboard, I think that OpenCR is enough because 5V is supplied. You can make the above cable for Upboard yourself.

florianspy, 2018-05-23:
Thank you so much for your help ! 
I will try out all your solutions as soon as possible.

robotpilot, 2018-05-23:
@florianspy
Thank you for your consideration and supports.
These questions have increased in recent years. We will explain how to change the SBC and sensors in the manual ASAP.

robotpilot, 2018-05-25:
Hi, again @florianspy!

We update and release new firmware (version 1.2.0) and new software (version 1.0.0) for users of TurtleBot3. :) 

Also, I added a description of the SBC replacement you mentioned on e-manual below.

http://emanual.robotis.com/docs/en/platform/turtlebot3/compatible_devices/#compatible-devices

Thank you to your contribution!

Please refer to the link below for details of this update.

https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888

closed 2018-05-25 by robotpilot