ROBOTIS-GIT/turtlebot3 #203: Robot model on RVIZ displays itself as backwards to that of the actual robot

MuratCA77, 2018-05-12:
ISSUE TEMPLATE ver. 0.0.1

1. Which TurtleBot3 you have?

- [x] burger
- [ ] waffle
- [ ] waffle pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

- [x] Raspberry Pi 3
- [ ] Intel Joule
- [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

- [x] Ubuntu MATE 16.04 or later
- [ ] Raspbian
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in RemotePC?

- [x] Ubuntu 16.04 LTS (Xenial Xerus)
- [ ] Ubuntu 18.04 LTS (Bionic Beaver)
- [ ] Linux Mint 18.X
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

 - Software version : [0.1.2]
 - Firmware version : [1.1.2]
 
4. Write down the commands you used in order

 [RemotePC] roscore
 [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
 [RemotePC] roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
 
5. Copy and Paste your error message on terminal

N/A

6. Please, describe detailedly what difficulty you are in 

The turtlebot robot model on RVIZ displays itself as backwards to that of the actual robot such that when I use keyboard teleop to control, the actual robot moves forward but RVIZ displays it moving backwards. Changing the teleop commands does not work as it still ends up going in opposite directions

routiful, 2018-05-14:
Hi :)

Would you check Dynamixel ID setup? The left motor is set `1` and right motor is set `2`. You can see how to check it on [FAQ](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#turtlebot3-setup-dynamixels) in Wiki.

Thanks

MuratCA77, 2018-05-14:
I thought the Dynamixel actuators used in the TurtleBot3 have a factory set IDs and configurations??

MuratCA77, 2018-05-14:
Hello ROBOTIS Team,

I have checked both motors and here are the results:

LEFT MOTOR

Succeeded to open the port! 
 - Device Name : /dev/OpenCR  
 - Baudrate    : 57600  
 
Start turtlebot3 setup motor 
  
  
1. setup left  motor 
2. setup right motor 
3. test  left  motor 
4. test  right motor 
>> Do you really want to setup ? y/n : yes 
setup.... left 
Find Motor... 
    setbaud : 57600  
[TxRxResult] There is no status packet! 
    setbaud : 1000000  
    ... SUCCESS 
    [ID:1] 
    found motor 
Setup Motor Left... 
    ok 
  
  
1. setup left  motor 
2. setup right motor 
3. test  left  motor 
4. test  right motor 
>> 

RIGHT MOTOR

1. setup left  motor 
2. setup right motor 
3. test  left  motor 
4. test  right motor 
>> Do you really want to setup ? y/n : yes 
setup.... right 
Find Motor... 
    setbaud : 57600  
[TxRxResult] There is no status packet! 
    setbaud : 1000000  
    ... SUCCESS 
    [ID:2] 
    found motor 
Setup Motor Right... 
    ok 
  
  
1. setup left  motor 
2. setup right motor 
3. test  left  motor 
4. test  right motor 
>> 

MOTOR TESTS

1. setup left  motor 
2. setup right motor 
3. test  left  motor 
4. test  right motor 
>> test.... right 
Test Motor Left... 
  
  
1. setup left  motor 
2. setup right motor 
3. test  left  motor 
4. test  right motor 
>> test.... left 
Test Motor Left... 
  
  
1. setup left  motor 
2. setup right motor 
3. test  left  motor 
4. test  right motor 

I was able to move both wheels clockwise only during testing.

MuratCA77, 2018-05-14:
Now, this is getting very frustrating because after doing what I was told about dynamixels, it says "unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

MuratCA77, 2018-05-14:
I have not fixed the sync issue by uploading turtlebot_core again. But this time, motors are not even moving when I use teleop. This is getting very frustrating

MuratCA77, 2018-05-14:
I meant to say I fixed the sync issue in my previous message.

kijongGil, 2018-05-15:
Hi @MuratCA77 :)
First time when I see your issue, I doubted the LDS had been assembled in the opposite direction.
But now, I doubted that you have assembled the left and the right of the motor in reverse.
When you press 'w', the actual robot moves forward? If so, change Dynamixel ID of left and right.

Thanks.

kijongGil, 2018-05-15:
No, your Dynamixel have assembled well.
Try to run again on gazebo without the actual turtlebot3.

1. Run the empty world in gazebo
```
export TURTLEBOT3_MODEL=burger
roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch
```
You can see a turtlebot3 in gazebo.

2. Run remote.launch
`roslaunch turtlebot3_bringup turtlebot3_remote.launch`

3. Run rviz
`rosrun rviz rviz`

4. Run teleop.launch
` roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch`

please tell me that turtlebot3 in gazebo moves right direction or not.

Thanks.

MuratCA77, 2018-05-16:
Hi, Murat

In video, the turtlebot3 is perfectly working  well. When you press 'w',
turtlebot3 in gazebo moves forward.
But, you said that when you press 'w', actual turtlebot3 moves forward.

Have you confused the front and back of the turtlebot3?

2018-05-16 1:24 GMT+09:00 Aksu, Murat (Fed) <murat.aksu@nist.gov>:

> Hello Gilbert,
>
>
>
> Here is the video demo for your instructions. My first command was “w’ and
> then second command was “x” for the simulation.
>
>
>
> Thanks,
>
> Murat
>
>
>
> *From:* Gilbert [mailto:notifications@github.com]
> *Sent:* Tuesday, May 15, 2018 11:57 AM
> *To:* ROBOTIS-GIT/turtlebot3 <turtlebot3@noreply.github.com>
> *Cc:* Aksu, Murat (Fed) <murat.aksu@nist.gov>; Mention <
> mention@noreply.github.com>
> *Subject:* Re: [ROBOTIS-GIT/turtlebot3] Robot model on RVIZ displays
> itself as backwards to that of the actual robot (#203)
>
>
>
> No, your Dynamixel have assembled well.
> Try to run again on gazebo without the actual turtlebot3.
>
>    1. Run the empty world in gazebo
>
> export TURTLEBOT3_MODEL=burger
>
> roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch
>
> You can see a turtlebot3 in gazebo.
>
>    1. Run remote.launch
>    roslaunch turtlebot3_bringup turtlebot3_remote.launch
>    2. Run rviz
>    rosrun rviz rviz
>    3. Run teleop.launch
>    roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
>
> please tell me that turtlebot3 in gazebo moves right direction or not.
>
> Thanks.
>
> <https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FROBOTIS-GIT%2Fturtlebot3%2Fissues%2F203%23issuecomment-389219645&data=02%7C01%7Cmurat.aksu%40nist.gov%7Cfd66d71327734a3ded9108d5ba7c8aed%7C2ab5d82fd8fa4797a93e054655c61dec%7C1%7C0%7C636619966428655503&sdata=y4OEnklsXG4qtODjXCgHsFWCpKPfPioedzg5Dg4U4gU%3D&reserved=0>,
> .[image: Image removed by sender.]

-- 
-- 

Kijong Gil (kkjong@robotis.com)
Researcher / R&D Dept. OpenSourceTeam

ROBOTIS CO., LTD. (http://www.robotis.com)
#1505 Ace High End Tower NO. 3,
Gasan-dong 145
<https://maps.google.com/?q=Gasan-dong+145&entry=gmail&source=g>, GasanDigital
1-ro,
GeumCheon-gu, Seoul, Korea 153-787
Tel : +82-70-8671-2841, Fax: +82-70-8230-xxxx

Follow ROBOTIS on Twitter <https://twitter.com/robotis>, Facebook
<https://www.facebook.com/robotis.company>, Linkedin
<http://www.linkedin.com/company/robotis>

MuratCA77, 2018-05-16:
Dear Gilbert,

When I press ‘w’, the robot in Gazebo moves backwards. When I press ‘x’, the robot in Gazebo moves forward. Is that how it is supposed to be? I do apologize for confusion.

But when it comes to control the actual robot with teleop, I am seeing completely opposite of what I am seeing in Gazebo. I think the following link can explain my issue better.

https://answers.ros.org/question/253397/turtlebot-rviz-visualisation-opposite-from-real-robot/

Best,
Murat

2018-05-16 1:24 GMT+09:00 Aksu, Murat (Fed) <murat.aksu@nist.gov<mailto:murat.aksu@nist.gov>>:

Kijong Gil (kkjong@robotis.com<mailto:kkjong@robotis.com>)

robotpilot, 2018-05-20:
Here are the configuration on URDF we use. I hope this helps you.
(TurtleBot3 Burger with Raspberry Pi Camera)
https://github.com/ROBOTIS-GIT/turtlebot3/blob/develop/turtlebot3_description/urdf/turtlebot3_burger_for_autorace.gazebo.xacro#L134-L165
(TurtleBot3 Waffle Pi with Raspberry Pi Camera)
https://github.com/ROBOTIS-GIT/turtlebot3/blob/develop/turtlebot3_description/urdf/turtlebot3_waffle_pi.gazebo.xacro#L143-L175

standin000, 2018-05-21:
@MuratCA77 I have the same issue about right motor like you, how do you fix it? thanks!

robotpilot, 2018-05-23:
The original issue seems to have been resolved. If you have other issues, please open a new issue. I will close this issue. Thanks.

robotpilot, 2018-05-23:
@MuratCA77, If this issue has not been resolved yet, please open a new issue. Thanks.

closed 2018-05-23 by robotpilot