ROBOTIS-GIT/turtlebot3 #204: How to use large capacity battery ？

ncnynl, 2018-05-15:
The battery capacity is 8800mah, but more than 2 hours, burger will always sound, but the actual battery has power

routiful, 2018-05-16:
TB3 makes warning sounds when voltage of the battery is drop under 11.0v.

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/src/turtlebot3/turtlebot3_diagnosis.cpp#L134

So it may actual battery's power is on, you can hear the sound always.

Thanks.

routiful, 2018-05-16:
And next time, Please FOLLOW ISSUE TEMPLATE. if not, we are going to close the issue without any comment.

ncnynl, 2018-05-16:
@routiful    Thank you ,  I will check it again .

robotpilot, 2018-05-20:
This issue is closed because it is covered in issue #205.
Thank you.

closed 2018-05-20 by robotpilot