ROBOTIS-GIT/turtlebot3 #205: 1800mah

ncnynl, 2018-05-15:


routiful, 2018-05-16:
And next time, Please FOLLOW ISSUE TEMPLATE. if not, we are going to close the issue without any comment.

ncnynl, 2018-05-16:
@routiful  sorry,  I supplement relevant content

robotpilot, 2018-05-17:
@OpusK, have you seen this problem in the past?

ncnynl, 2018-05-17:
@OpusK    the battery: 
 LI-PO 11.1 v 1800 mAh LB -12 19.98wh  5C (473474P 4HR5C)   
Made in china  Q.C.Pass : 17/45

OpusK, 2018-05-18:
Does OpenCR not work when powered on with 12V adapter?
Have you measured 11.1V, 5V, 3.3V of OpenCR?
And what about the LED status of the board?
Please manually enter the Bootloader and check that the "STATUS" LED is blinking.

ncnynl, 2018-05-18:
@OpusK   After the car not work properly Suddenly  .  the openCR can not work again .   no matter powered on  with 12V adapter or  battery .  Use a new OpenCR , test with 12V adapter is ok, But test with battery.  it can not work .  test with 12V adapter  again , it can not work too .

OpusK, 2018-05-21:
Please check the fuse of OpenCR which does not work.

[image]

If the fuse is normal, both voltages will be normal, but if the fuse is broken, only one side will measure the voltage. 

And, please check again that the electrode of the battery matches the electrode of the board connector.

ncnynl, 2018-05-23:
@OpusK  I will check it ， thank you for your patient

robotpilot, 2018-05-27:
Hi @ncnynl, @OpusK,
Has this problem been resolved?

ncnynl, 2018-05-28:
@robotpilot  not yet , I am busy for other things , i will check the fuse  soon.  thank you.