ROBOTIS-GIT/turtlebot3 #207: Multiple TurtleBot3's

bsb808, 2018-05-16:
Hello,

I am attempting to work with two TurtleBot3 platforms simultaneously.  For the ROS topics/services, using a namespace works just fine with small edits to the bringup launch file.

The problem I have is how to add a prefix to the TF broadcasting done by the turtlebot3_core node.  It appears that the TF determination is done on the OpenCR board and then sent over serial using rosserial_python/serial_node.py.

I did try adding a "tf_prefix" parameter definition, but it appears that this is not implemented for the OpenCR/rosserial_python solution.

Any ideas on how to add distinct TF prefixes to each robot (e.g., "robot1/odom" and "robot2/odom", etc. for two robots)?

The only thing I can think of is to add a #define to the [turtlebot3_core_config.h](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core_config.h), but that would require recompiling the firmware for each change, which doesn't seem like a good solution.

Thank you,
Brian

ISSUE TEMPLATE ver. 0.0.1

1. Which TurtleBot3 you have?

- [X] burger
- [ ] waffle
- [ ] waffle pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

- [X] Raspberry Pi 3
- [ ] Intel Joule
- [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

- [X] Ubuntu MATE 16.04 or later
- [ ] Raspbian
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in RemotePC?

- [X] Ubuntu 16.04 LTS (Xenial Xerus)
- [ ] Ubuntu 18.04 LTS (Bionic Beaver)
- [ ] Linux Mint 18.X
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

 - Software version : [0.1.2]
 - Firmware version : [1.1.2]

robotpilot, 2018-05-18:
Hi @bsb808,

I apologize for the delay and regret any inconvenience it may have caused.

Since ROS 1.x is not a system made by thinking about multi-robot, tf is somehow tied in one common way. This makes it difficult to create a multi-robot system. The ROS community has improved on ROS2 about it, but now we are using ROS 1.x, so we have to consider a roundabout way.

Both of the solutions you mentioned are possible. 
But you too, I just want to remap it in the launch file. Unfortunately, this is impossible.

In Chapter 12 of the following ROS book, we introduced three service robots using TurtleBot3.
At this time, we also did similar things to you, but eventually, we fixed it by modifying our OpenCR firmware.
https://community.robotsource.org/t/download-the-ros-robot-programming-book-for-free/51

So, I recommend you to modify the OpenCR firmware.

I recently received the demand from many people interested in multi-robot using TurtleBot3.I will look for a better way of doing this.

I appreciate for your questions and suggestions.

Best regards,
Pyo

bsb808, 2018-05-18:
Thank you for the very helpful reply.   I am surprised that it isn't a solved issue. 
 I'll start working on the option.  If I can come up with something efficient I'll make sure and post it for others.

Thank you again,
Brian

ncnynl, 2018-05-18:
@bsb808  I also work on it but not done . by modifying the OpenCR firmware.

bsb808, 2018-05-18:
Any chance you would be willing to share the modification?

robotpilot, 2018-05-20:
We will gladly share if there is something that we would most recommend among the many ways.

routiful, 2018-07-12:
Hello :)

We updated new firmware (v1.2.1) and software today.
This update includes more simple way to load multiple turtlebot3.

You can get more information about it in [ROS Discourse](https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888/6) and [WiKi](http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#load-multiple-turtlebot3s)

Thank you for your inquiry and suggestions!

bsb808, 2018-07-12:
Many thanks!  That is great news.  I look forward to giving it a try!
Brian

robotpilot, 2018-07-12:
Hi @bsb808,
We were able to solve this problem with your help. thank you very much.
Once again, thank you for your help and supports. :)

closed 2018-07-12 by robotpilot

wx0205, 2018-12-13:
Hello, now I make a project about navigation with multi-turtlebot3 in real Environment. However, when i add  the namespace for each turtlebot3, it can locate the initial pose. Could you can help me? Thank you!