ROBOTIS-GIT/turtlebot3 #208: raspberry pi Ubuntu MATE turtlebot3 package catkin_make freezing problem.

JeenQ, 2018-05-20:
ISSUE TEMPLATE ver. 0.0.1

1. Which TurtleBot3 you have?

- [  ] burger
- [ ] waffle
- [ ] waffle pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

- [  ] Raspberry Pi 3
- [ ] Intel Joule
- [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

- [  ] Ubuntu MATE 16.04 or later
- [ ] Raspbian
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in RemotePC?

- [  ] Ubuntu 16.04 LTS (Xenial Xerus)
- [ ] Ubuntu 18.04 LTS (Bionic Beaver)
- [ ] Linux Mint 18.X
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

 - Software version : [0.1.2]
 - Firmware version : [1.1.3]
 
4. Write down the commands you used in order

 [SBC] 
$ cd ~/catkin_ws/src
$ git clone https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver.git
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3.git
$ cd ~/catkin_ws && catkin_make
 
5. Copy and Paste your error message on terminal

6. Please, describe detailedly what difficulty you are in 
After "catkin_make" on terminal, hls_lfcd_lds_driver package and turtlebot3_msgs package are okay, but when turtlebot3 package starts building, after a while the raspberry pi(ubuntu MATE) freezes. I'm using remote desktop to use raspberry pi. In a few minutes, the remote desktop disconnected, and I can't connect anymore before I reboot the raspberry pi. I found my own solution by using "catkin_make_isolated" instead of "catkin_make". The build is successful with the "catkin_make_isolated". Is this problem occurs only on me? I frequently face this problem.

robotpilot, 2018-05-20:
Hi @JeenQ 

1. Did you delete some packages that are not needed in TurtleBot SBC?
http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#install-dependent-packages
```
$ cd ~/catkin_ws/src/turtlebot3
$ sudo rm -r turtlebot3_description/ turtlebot3_teleop/ turtlebot3_navigation/ turtlebot3_slam/ turtlebot3_example/
```

2. If you stop during 'catkin_make', I recommend using the command 'catkin_make -j1'.
```
$ cd ~/catkin_ws && catkin_make -j1
```
This is one option of the 'make' command: 
"-j [N], --jobs [= N] Allow N jobs at once; infinite jobs with no arg".
If you set the option to '-j1', only one process will be used when building.

JeenQ, 2018-05-21:
I was expecting the reason was the -j4 option.
It seems multithread job is a big burden for raspberry pi.
It works okay. Thanks @robotpilot !

closed 2018-05-21 by JeenQ