ROBOTIS-GIT/turtlebot3 #209: Unable to build raspicam_node and unable to sudo apt-get ros-kinetic-...

jball11, 2018-05-21:
ISSUE TEMPLATE ver. 0.0.1

1. Which TurtleBot3 you have?

- [ ] burger
- [ ] waffle
- [x] waffle pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

- [x] Raspberry Pi 3
- [ ] Intel Joule
- [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

- [ ] Ubuntu MATE 16.04 or later
- [x] Raspbian
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in RemotePC?

- [x] Ubuntu 16.04 LTS (Xenial Xerus)
- [ ] Ubuntu 18.04 LTS (Bionic Beaver)
- [ ] Linux Mint 18.X
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)
 
5. Copy and Paste your error message on terminal

**Error for 26.5.1 of the turtlebot3 e-manual:**

```
pi@raspberrypi:~/catkin_ws/src $ sudo apt-get install ros-kinetic-compressed-image-transport ros-kinetic-camera-info-manager
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-compressed-image-transport
E: Unable to locate package ros-kinetic-camera-info-manager
```

**Error for raspicam_node build instructions:**

```
pi@raspberrypi:~/catkin_ws $ rosdep install --from-paths src --ignore-src --rosdistro=kineticERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
turtlebot3_bringup: No definition of [turtlebot3_teleop] for OS version [stretch]
turtlebot3: No definition of [turtlebot3_teleop] for OS version [stretch]
```

SUMMARY
========
  
6. Please, describe detailedly what difficulty you are in 

When i attempt 26.5.1 of the turtlebot3 e-manual i receive an error. When i attempt the raspicam_node instructions i also receive an error. I attempting both procedures in my raspberry pi terminal. I have followed all previous steps of 26.5.1 and raspicam_node build instructions with success multiple times but when it comes to the specified lines of code above i receive the attached errors.

robotpilot, 2018-05-22:
@ jball11

I'm sorry for the inconvenience.

1. The contents 26.5.1 of the TurtleBot3 e-manual are for Ubuntu MATE 16.04 users. There is no mention of OS. I'm sorry. I'll fix it.

2. The ROS community does not provide a binary ROS package for Raspbian, so you can't install it with the apt-get install command. To do this, we provide a Raspbian image for TurtleBot3.
The Raspbian image below contains the raspicam_node package you are installing.

http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#install-linux-based-on-raspbian

jball11, 2018-05-22:
@robotpilot 

Thanks for the reply.

What do i need to do to get my raspberry pi cam to work on ROS?

robotpilot, 2018-05-23:
@jball11

Thanks for your interest in TurtleBot3. 
We support you. Have a good day. :)

closed 2018-05-23 by robotpilot