ROBOTIS-GIT/turtlebot3 #217: cartographer_node is not publishing 'map' tf frame. Fixed frame map does not exist.

YuehChuan, 2018-05-28:
ISSUE TEMPLATE ver. 0.0.1

1. Which TurtleBot3 you have?

- [x] burger
- [ ] waffle
- [ ] waffle pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

- [x] Raspberry Pi 3
- [ ] Intel Joule
- [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

- [x] Ubuntu MATE 16.04 or later
- [ ] Raspbian
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in RemotePC?

- [x] Ubuntu 16.04 LTS (Xenial Xerus)
- [ ] Ubuntu 18.04 LTS (Bionic Beaver)
- [ ] Linux Mint 18.X
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

 - Software version : [0.1.2]
 - Firmware version : [1.1.2]
 
4. Write down the commands you used in order

 [RemotePC] roscore
 [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
 [RemotePC] roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=cartographer

 
5. Copy and Paste your error message on terminal

after launch cartographer  it said no map tf publish 
https://asciinema.org/a/tLSrUgDJCbJcnFlXRI0SHN7PX
so I force it to publish map 
with 
`rosrun tf static_transform_publisher 0 0 0 0 0 0  world map 10`
got error. 

I think it is relate to issue 
cartographer_node is not publishing 'map' tf frame. Fixed frame map does not exist.
googlecartographer/cartographer_ros#300
googlecartographer/cartographer_ros#202

I guess it's  .lua configuration , not yet looks into details. 
```
[ WARN] [1501071021.570063704]: W0726 20:10:21.000000 14280 tf_bridge.cc:51] Invalid argument passed to lookupTransform argument source_frame in tf2 frame_ids cannot be empty
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 122 in /tmp/binarydeb/ros-kinetic-tf2-0.5.16/src/buffer_core.cpp
^CWarning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 122 in /tmp/binarydeb/ros-kinetic-tf2-0.5.16/src/buffer_core.cpp
```
  
6. Please, describe detailedly what difficulty you are in 

cartographer_node is not publishing 'map' tf frame. Fixed frame map does not exist.
TF terminology issue?

YuehChuan, 2018-05-29:
@robotpilot  Thanks for the detail TF and rqt_graph.
I build from source with above steps. 
It's  long time ago, so I direct  git pull without checking the version. 
Looks it works fine,  I would try it with  0.2.0 here.
https://github.com/googlecartographer/cartographer/releases
TQ!

YuehChuan, 2018-05-29:
[image]
oh, 0.2.0 has no install_proto3.sh
However I would like to make latest cartographer work on kinetic. 
I will close the issue for now and  If it works I will update here.

YuehChuan, 2018-05-29:
Ooops get no luck with 0.2.0 
https://asciinema.org/a/ud06KuXElX8nKgSykbsdqdiRZ
and there is no cartographer_occupancy_grid_node in 0.2.0

download .zip file from 
https://github.com/googlecartographer/cartographer/releases
https://github.com/googlecartographer/cartographer_ros/releases
put it under catkin_ws/src

```txt=
# Install wstool and rosdep.
sudo apt-get update
sudo apt-get install -y python-wstool python-rosdep ninja-build
# Create a new workspace in 'catkin_ws'.
mkdir catkin_ws
cd catkin_ws
wstool init src
# Install deb dependencies.
rosdep init
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
# Build and install.
catkin_make_isolated --install --use-ninja
source install_isolated/setup.bash
```

list issue related
https://github.com/googlecartographer/cartographer/issues/765

robotpilot, 2018-05-29:
It seems to have misunderstood. 
I have cloned the cartographer and cartographer_ros master branch, built it from source code, and used it in ROS Kinetic.

YuehChuan, 2018-05-30:
[image]
thanks for the update! 
I wonder whether the software on my pi3  is obsolete

robotpilot, 2018-05-31:
Huh. I think it is strange now. 
How about organizing it using the command below?

```bash
sudo apt update && sudo apt upgrade && sudo apt autoremove
```

YuehChuan, 2018-06-04:
I just fixed it yesterday! Thanks!
[image]

YuehChuan, 2018-06-04:
However, I found a further interesting issue, I will discribe below, and how I fixed it.

YuehChuan, 2018-06-04:
the way how  I resolve the issue, and interesting discover about cartographer sensor delivered data out of order.
https://hackmd.io/s/r1fAEBGlX#

robotpilot, 2018-06-05:
Hi @YuehChuan,
This is an issue that we have not seen before.
I will close this issue and open a new issue thread. Thanks!

closed 2018-06-05 by robotpilot