ROBOTIS-GIT/turtlebot3 #221: Cannot read sensor state information since the new update

masacha, 2018-05-29:
ISSUE TEMPLATE ver. 0.0.1

1. Which TurtleBot3 you have?

- [x ] burger
- [ ] waffle
- [ ] waffle pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

- [ x] Raspberry Pi 3
- [ ] Intel Joule
- [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

- [ x] Ubuntu MATE 16.04 or later
- [ ] Raspbian
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in RemotePC?

- [ x] Ubuntu 16.04 LTS (Xenial Xerus)
- [ ] Ubuntu 18.04 LTS (Bionic Beaver)
- [ ] Linux Mint 18.X
- [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

 - Software version : [1.0.0]
 - Firmware version : [1.2.0]
 
4. Write down the commands you used in order

 [RemotePC] roscore
 [SBC] roslaunch turtlebot3_bringup turtlebot3_core.launch
 [RemotePC] roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
[RemotePC] rqt
 
5. Copy and Paste your error message on terminal

No error on terminal. 

6. Please, describe detailedly what difficulty you are in 

Since a few days, I can't tele-operate Turtlebot3 burger. No error appears on the terminal, the motors simply don't move. When I open rqt, the sensor_state topic displays unknown bandwidth and unknown frequency. 

When I use a custom code for OpenCR (similar to the Turtlebot3 Segway one) with pwm control, I can control the motors but the sensor_state remains unknown. Therefore I cannot access encoder values. 

The same problem happens on two Turtlebots.

I tried to do Dynamixel motor setup with one motor connected (and powed on), but this error is displayed:
Find Motor...
    setbaud : 57600

[TxRxResult] There is no status packet!
    setbaud : 1000000

[TxRxResult] There is no status packet!

What do you recommend I do? Should I downgrade to a previous software and firmware version?

routiful, 2018-05-29:
Hi :)

I assumed Dynamixel has broken since unknown error. 
You can try firmware recovery by using [RoboPlus Manager 2.0](http://emanual.robotis.com/docs/en/software/rplus2/manager/)

After uploading firmware to Dynamixel, attempt to motor setup and launch teleop. 

Thanks

masacha, 2018-05-29:
Hi! :) 
Thanks for the reply. 
Ok, I will attempt to make a firmware recovery of Dynamixel. To use RoboPlus Manager, I need to connect Dynamixel to PC with a connector (for instance U2D2), is that right?

routiful, 2018-05-29:
That's right, Or else you can use OpenCR which is uploaded [usb_to_dxl](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/usb_to_dxl/usb_to_dxl.ino) as U2D2.

masacha, 2018-06-04:
Thanks for your help, I use several Turtlebot3 for academic purposes in my university laboratory and I really appreciate your reactivity. 

I have tried the same motors with another OpenCR board and strangely, everything works well. So maybe the Dynamixel is not broken.

I now have two OpenCR boards that can read the sensor_state from the motors and two OpenCR boards that cannot (with the same two motors). Is it a problem from the OpenCR or is it still a problem from the Dynamixel?

routiful, 2018-06-04:
Hi :)

It looks like problem from the OpenCR. Then, please try re-update bootloader to OpenCR and upload turtlebot_core (v.1.2.0).
After that, if the problem is not solved, please contact ROBOTIS customer support team (contactus2@robotis.com)

Thanks

masacha, 2018-06-06:
Thank you for your reply! I re-burned the bootloader and updated the latest turtlebot3_core on OpenCR but still, the sensor_state topic is not published correctly (unknown bandwith and unknown frequency). It looks a lot like this issue: https://github.com/ROBOTIS-GIT/turtlebot3/issues/162 but I don't know if it is the same case.  
Should I send the two OpenCR boards with this issue for repair?

routiful, 2018-06-07:
Yes, it might be hardware fault from OpenCR. Please contact ROBOTIS customer support team. Then they are going to inform you where to send yours.

Sorry for inconveniences

Thanks

masacha, 2018-06-07:
I will do that, thank you very much!

routiful, 2018-06-07:
Great.  If you will face similar problem when you replace OpenCR, you would re-open this issue.

Thanks :)

closed 2018-06-07 by routiful