ROBOTIS-GIT/turtlebot3 #228: IMU data time on Cartographer

robotpilot, 2018-06-05:
about sensor delivered data out of order.
https://hackmd.io/s/r1fAEBGlX#

YuehChuan, 2018-06-08:
Thanks for openning this!

routiful, 2018-07-12:
Hello @YuehChuan :)

We updated new firmware (v1.2.1) and software today
This update includes modified time function in roslib and amended config file for cartographer

After fixed it, the cartographer was not crashed with IMU and without IMU for a long time.

You can get more information about it in [ROS Discourse](https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888/6) 

Thank you for your inquiry and suggestions!

robotpilot, 2018-07-12:
Hi @YuehChuan,
Thank you for your consideration and supports.
With your help, we were able to find this issue early and to improve it and release it.
I close this issue. You can reopen this issue to show this issue to the users whenever. 
Once again, thank you for your help and supports. :)

closed 2018-07-12 by robotpilot