ROBOTIS-GIT/turtlebot3 #229: Problem with rviz : No transform from [wheel_left/right_link] to [base_footprint]

nns431, 2018-06-05:
Hi, I'm very new here and I got a problem.
I follow the TurtleBot3 e-manual from the start and stuck in step 7.Bringup
This is what I'm using now:
- **TurtleBot3** Burger with Raspberry Pi 3, Ubuntu MATE 16.04
- **RemotePC** 8GB RAM, Intel® Core™ i7-5500U, Intel® HD Graphics 5500 on board (with addition AMD card), 64-bit, OS Ubuntu 16.04 LTS (dual with Windows10)

Here is **command order**
    [Remote PC] roscore
    [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
    [Remote PC] export TURTLEBOT3_MODEL=burger
                          roslaunch turtlebot3_bringup turtlebot3_remote.launch
                          rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz
 
There is no error message on terminal.
The problem in the picture below
[image]

Further more, I try step 8.1 Topic Monitor there are more problem here like this
[image]

I have searched and try many solution but it didn't work.
I have already check the hardware, it's connected correctly and work properly.

Many thanks,

kijongGil, 2018-06-05:
Hi, @nns431 :)
I think that the left and right DXL's ID was wrong setup. First, try to test your DXL is working. refer to this
http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#basic-operation
And then, setup to DXL's ID (left : 1, right : 2) using R+ manager 2.0.
http://emanual.robotis.com/docs/en/software/rplus2/manager/#roboplus-manager-20

Thanks,
Gilbert

nns431, 2018-06-05:
Thanks for your reply.
I have updated dynamixel firmware & check the ID.
It's now up to date & have correct ID, but the same problem.

nns431, 2018-06-05:
I try more by change the OpenCR board to check if the first one is broken, but same thing occur with the second one. I either try to swap the dynamixel wire connect to OpenCR board but it doen't work too.
So, I have no idea where is the problem comes.

kijongGil, 2018-06-07:
Please check your OpenCR version 1.3.0 and turtlebot3 packages.
Try again check topic monitor, and did you set up network configuration? refer to link http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#5-network-configuration

nns431, 2018-06-07:
I'm just found why it's not working.
It because I set both Host&Master IP with the same IP from TurtleBot.
How stupid I am !! I just lost 2 days because wrong IP hahaha.
I'm now closing this issue. I'm feel so thankful for your help @kijongGil .

closed 2018-06-07 by nns431

kijongGil, 2018-06-07:
I am glad that the problem has been solved.
Please, let me know if you have any questions on this issues.

Thanks,
Gilbert

ubtjoe, 2019-06-28:
@nns431 Please provide detailed solution.  I am having the same problem, but your description of the solution is not clear to me.  On which machine (turtlebot or remote_pc) do you have the Host&Master IP the same?  Thanks!