ROBOTIS-GIT/turtlebot3 #230: TurtleBot3 LDS-01

garinalham, 2018-06-05:
Hi, I'm totally new to electronics and programming. My question is, is there any way to change pwm and/or the rpm of the LIDAR from the TurtleBot3 kit? Thank you.

kijongGil, 2018-06-05:
Hi, @garinalham :)
Unfortunately, the speed of the LDS-01 can not be changed. Because this function is in firmware.
Could you tell me that why you want to change the speed of the LDS? I will help you if I can.

garinalham, 2018-06-05:
Thanks for the respond! Yeah, I need to see how different scanrate affect the mapping result. Or do you have any alternative to this? Or should I just update the map_update_interval?

kijongGil, 2018-06-05:
Yes... To control the map_update_interval is best way. But, now I have a idea. 
How about change map_metadata topic rate? This link will help you.
http://wiki.ros.org/gmapping

Thanks,
Gilbert

garinalham, 2018-06-05:
Thanks! That was quick :). Can you elaborate what parameters is there in the map_metadata to tinker with? Perhaps a close replacement to the LIDARS rotational speed.

kijongGil, 2018-06-07:
I am glad that the problem has been solved.
Please, let me know if you have any questions on this issues.

Thanks,
Gilbert

closed 2018-06-11 by robotpilot