ROBOTIS-GIT/turtlebot3 #231: Problem with running karto slam method

KarloLiovic, 2018-06-05:
ISSUE TEMPLATE ver. 0.1.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [x] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
4. Write down the commands you used in order

    [Remote PC] roscore
    [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
    [Remote PC] roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=karto
 
5. Copy and Paste your error message on terminal

    ... logging to /home/karlo/.ros/log/d775c8da-68a3-11e8-9d15-1c659dfd216b/roslaunch-karlo-14305.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.129.2.2:44615/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /slam_karto/angle_variance_penalty: 0.349
 * /slam_karto/coarse_angle_resolution: 0.0349
 * /slam_karto/coarse_search_angle_offset: 0.349
 * /slam_karto/correlation_search_space_dimension: 0.3
 * /slam_karto/correlation_search_space_resolution: 0.01
 * /slam_karto/correlation_search_space_smear_deviation: 0.03
 * /slam_karto/distance_variance_penalty: 0.3
 * /slam_karto/do_loop_closing: True
 * /slam_karto/fine_search_angle_offset: 0.00349
 * /slam_karto/link_match_minimum_response_fine: 0.12
 * /slam_karto/link_scan_maximum_distance: 3.5
 * /slam_karto/loop_match_maximum_variance_coarse: 0.4
 * /slam_karto/loop_match_minimum_chain_size: 10
 * /slam_karto/loop_match_minimum_response_coarse: 0.8
 * /slam_karto/loop_match_minimum_response_fine: 0.8
 * /slam_karto/loop_search_maximum_distance: 3.5
 * /slam_karto/loop_search_space_dimension: 8.0
 * /slam_karto/loop_search_space_resolution: 0.05
 * /slam_karto/loop_search_space_smear_deviation: 0.03
 * /slam_karto/minimum_angle_penalty: 0.9
 * /slam_karto/minimum_distance_penalty: 0.5
 * /slam_karto/minimum_travel_distance: 0.12
 * /slam_karto/minimum_travel_heading: 0.174
 * /slam_karto/scan_buffer_maximum_scan_distance: 3.5
 * /slam_karto/scan_buffer_size: 70
 * /slam_karto/use_response_expansion: False
 * /slam_karto/use_scan_barycenter: True
 * /slam_karto/use_scan_matching: True

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    slam_karto (slam_karto/slam_karto)

ROS_MASTER_URI=http://10.129.2.2:11311

core service [/rosout] found
process[robot_state_publisher-1]: started with pid [14326]
process[slam_karto-2]: started with pid [14327]
process[rviz-3]: started with pid [14328]
[ INFO] [1528193200.683855510]: laser base_scan's pose wrt base: -0.032 0.000 0.000
Info: clipped range threshold to be within minimum and maximum range!
Registering sensor: [base_scan]
LaserRangeScan contains 360 range readings, expected 361
LaserRangeScan contains 360 range readings, expected 361
LaserRangeScan contains 360 range readings, expected 361
LaserRangeScan contains 360 range readings, expected 361
LaserRangeScan contains 360 range readings, expected 361
LaserRangeScan contains 360 range readings, expected 361
LaserRangeScan contains 360 range readings, expected 361
LaserRangeScan contains 360 range readings, expected 361
LaserRangeScan contains 360 range readings, expected 361
...

  
6. Please, describe detailedly what difficulty you are in 

    I'm having issues with starting karto slam. I haven't been able to find a file where i can change the expected number of range readings.

routiful, 2018-06-08:
Hi :)

This problem is going to be solved to update `hls_lfcd_lds_driver` package.
Please refer below commands

[SBC]
$ cd ~/catkin_ws/src
$ cd hls_lfcd_lds_driver/
$ git pull
$ cd ~/catkin_ws && catkin_make -j1

or

[SBC]
$ cd ~/catkin_ws/src
$ rm -rf hls_lfcd_lds_driver/
$ git clone https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver.git
$ cd ~/catkin_ws && catkin_make -j1

Thanks

KarloLiovic, 2018-06-12:
That has solved the issue.

Thank you.

closed 2018-06-12 by KarloLiovic

saikishor, 2018-11-21:
@KarloLovic and @routiful That solution is only for your sensor right?. I have the similar issue using Velodyne. Any suggestions?.

LaserRangeScan contains 897 range readings, expected 899

EndlessLoops, 2021-06-11:
> @KarloLovic and @routiful That solution is only for your sensor right?. I have the similar issue using Velodyne. Any suggestions?.
> 
> LaserRangeScan contains 897 range readings, expected 899

Were you fix it ?