ROBOTIS-GIT/turtlebot3 #232: slam with 4_wheel TB3

idminerwilliam, 2018-06-07:
ISSUE TEMPLATE ver. 0.1.0

1. Which TurtleBot3 you have?

    Waffle pi model with 4 XL430-W250

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    Raspberry Pi 3

3. Which OS you installed in SBC?

    Raspbian

4. Which OS you installed in Remote PC?

    Ubuntu 16.04 LTS (Xenial Xerus)

3. Write down software version and firmware version

    ROS Packages for TurtleBot3 [1.0.0]
    OpenCR Board [1.2.0]
 
4. Write down the commands you used in order

   1. [Remote PC] roscore
   2. [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
   3. [Remote PC] export TURTLEBOT3_MODEL=waffle_pi
   4. [Remote PC] roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping
   5. [Remote PC] roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
  
5. Describe detailedly what difficulty you are in 

    I run slam in my office and control tb3 forward a few block and back.
    The map look not correct.
    
    [image]

    The door displacement ......
    I use four servos, so I only modify turtlebot3_motor_driver
    Maybe some setting need to modify too.
    How could I fix this slam problem ?

routiful, 2018-06-08:
Hi :)

You don't need to change any configuration for SLAM.

As you said, actual door is only one but the map created by your TB3 was showed pair? Do I understand properly?

Then, I assume that the corridor is too flat to make map. gmapping need to characteristic point as flowerpot, fire extinguisher. So put some object in the corridor before you make a map

Thanks

idminerwilliam, 2018-06-11:
Sorry, I tried to add some flowerpot, the map not good, too.
I found that my odom display in rviz not continuous, is that correct ？
[image]

routiful, 2018-06-12:
That's a weird. I don't know why the arrow is appeared spatially. Can you copy and paste a frequency of /odom topic ?
You can check below commands in terminal.

```
rostopic hz /odom
```

Furthermore, Odom calculation is based on two wheel. I think you have to change it depending your TB3.

https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_waffle/turtlebot3_core/turtlebot3_core.ino#L428

Thanks

idminerwilliam, 2018-06-12:
Hi routiful,
Thanks for your help.
I run `rostopic hz /odom`  when burger stop(i) & moving(ii).
i.
[image]
ii.
[image]

Maybe something was wrong when I modify calcOdometry in core.
I will try it !!

Thanks

routiful, 2018-06-14:
Furthermore, the message communication has problems. In general, the average rate is 25~30 hz.
Would you check connection between Raspberry pi 3 and RemotePC? 
I'll give you some references

FAQ : http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

Issue : #78