ROBOTIS-GIT/turtlebot3 #233: Elevate third layer of turtlebot3 waffle

florianspy, 2018-06-08:
ISSUE TEMPLATE ver. 0.1.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ x] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ x] etc (PLEASE, WRITE DOWN YOUR SBC HERE) Intel Upboard squared

3. Which OS you installed in SBC?

    - [ x] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
  
6. Please, describe detailedly what difficulty you are in 

   I want to elevated the third layer of the turtlebot 3 what files do i have to modify so that the data of the lidar sensor still is fine? Or is there no information stored regarding the coordinate system of the lidar sensor?

routiful, 2018-06-11:
Hi :)

You can find pose of the lidar sensor in urdf file. Follow attached link and modify it depending on your turtlebot3

https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_description/urdf/turtlebot3_waffle_pi.urdf.xacro#L160

Moreover, I give you a reference which is example of multiple layer

Hardware : http://emanual.robotis.com/docs/en/platform/turtlebot3/locomotion/#turtlebot3-friends-carrier

Software : https://github.com/ROBOTIS-GIT/turtlebot3_deliver

Thanks

florianspy, 2018-06-11:
Thank you alot for your fast and detailed answer!
So to get things right this file needs to be modified on the sbc as well as on the PC running roscore?

routiful, 2018-06-11:
No, you just modify URDF file in Remote PC. Because `robot_state_publisher` is launched in RemotePC.

http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup-a-turtlebot3-waffle-pi

florianspy, 2018-06-11:
Okay perfect thank you very much

routiful, 2018-06-11:
Nice, if you have any inquiry about it, please feel free to open issue :)

closed 2018-06-11 by routiful