ROBOTIS-GIT/turtlebot3 #237: roslaunch turtlebot3_bringup turtlebot3_robot.launch ERROR

oxidation99, 2018-06-16:
turtlebot3burger@turtlebot3burger-desktop:~$ roslaunch turtlebot3_bringup turtlebot3_lidar.launch
... logging to /home/turtlebot3burger/.ros/log/585c8fba-714b-11e8-ba6a-3010b3156803/roslaunch-turtlebot3burger-desktop-10242.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.200:44910/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.1.116:11311

Unable to launch [turtlebot3_lds-1]. 
If it is a script, you may be missing a '#!' declaration at the top.
The traceback for the exception was written to the log file
[turtlebot3_lds-1] killing on exit

How do I solve this problem?
I use turtlebot3 burger on Ubuntu mate.
Thank You 

[image]

robotpilot, 2018-06-17:
Hi @oxidation99 

If you want to launch Lidar sensor and core separately, It is correct to use the following command. However, if not for that purpose, it is redundant.
```sh
$ roslaunch turtlebot3_bringup turtlebot3_lidar.launch
$ roslaunch turtlebot3_bringup turtlebot3_core.launch
```
For TurtleBot3 bringup, the following command is sufficient. 
If you use it as a duplicate, it causes an error due to device duplication (/dev/ttyUSB0).
If you don't have a reason to run Lidar separately, I recommend running only the following command.
```sh
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
```

oxidation99, 2018-06-23:
Hi @robotpilot  thank you for your help.
I can run command roslaunch turtlebot3_bringup turtlebot3_robot.launch and separately two cammand.
I solved this by install ROS step by step on ROS/Tutorials website.
And now I have a question about LDS-01 I follow Appendixes LDS-01 but I can run this command ./lds_driver. How can I fix it?

robotpilot, 2018-06-24:
Hi @oxidation99,
The following information you mentioned is for general use, not ROS. First, for what purpose are you using it? And what error did you get?
http://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_lds_01/#user-guide-for-driver

closed 2018-07-02 by robotpilot