ROBOTIS-GIT/turtlebot3 #238: ERROR at roslaunch turtlebot3_bringup turtlebot3_remote.launch

oxidation99, 2018-06-17:
Hi 
When I use command roslaunch turtlebot3_bringup turtlebot3_remote.launch 
It not work as you can see at picture. I use turtlebot3 burger Ubuntu Mate 16.04.2(xenial) ROS kinetic on my pc I use ROS kinetic and install  Dependent ROS Packages.
How can I solved this problem.
Thank you

[image]

robotpilot, 2018-06-17:
Hi @oxidation99,

The execution of turtlebot3_remote.launch should be done on your Remote PC, not on your TurtleBot3 PC (Raspberry Pi 3).
http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup-a-turtlebot3-burger

oxidation99, 2018-06-19:
I run this command on my remote PC, not on Raspberry Pi and encounter the problem as pictured above.

kijongGil, 2018-06-19:
Hi @oxidation99 :)

Could you install ROS?
Run the roscore and tell me the result.

`$ roscore`

tunai, 2018-06-21:
Hi @oxidation99 

The launcher is looking for the model of your bot and failing to find it.

If this is the first time bringing up the robot, edit the "~/.bashrc" file on the remote PC and include this last line on it: “export TURTLEBOT3_MODEL = burger”. 

`$ sudo nano ~/.bashrc` (you can use your preferred text editor to do that, but let's consider the nano)

After the insertion of the new last line, hit CTRL+O to save the new file, and CTRL+X to exit (again, this is for the nano editor). Then, source the new bash file:

`$ source ~/.bashrc`

Now your workspace should know the model of your robot. **For every new terminal of the remote PC**, source the _devel_ folder in the catkin workspace:  

`$ source devel/setup.bash`

That should solve your problem.

oxidation99, 2018-06-23:
Thank you for all the help.
Hi @kijongGil  
I install step by step ROS based on ROS.org/Tutorials website.
This picture show when I run roscore.
[image]

And when I run command roslaunch turtlebot3_bringup turtlebot3_remote.launch after I run roscore command. This error show in picture two.

[image]

kijongGil, 2018-06-27:
Hi, @oxidation99. I'm sorry to reply late.
I think that your ROS installation have a problem.
Could you reinstall again?

closed 2018-07-02 by robotpilot