ROBOTIS-GIT/turtlebot3 #240: Can not upload to OpenCR

husikl, 2018-06-25:
Hello,

First of all thank you for the work you have done and you are doing, Turtlebot3 is a great project.

I am working on the Turtlebot3 waffle and I use Jetson TX2 as a SBC. 
We have recently purchased Open Manipulator arm for Turtlebot3.
I followed all the steps from the Emanual wiki to setup the hardware.

I have managed to set up Turtlebot3 and the robot was moving just fine, when I uploaded the turtlebot_core to the board, with OpenCR version 1.1.2.

The problem occured when I updated the version of OpenCR to 1.2.0 using Arduiono IDE to upload from examples turtlebot_with_open_manipulator_core to the board and test the dynamixel arm.
IDE is able to detect the board - dev/ttyACM0 and also I detect the board with libusb command.
When I upload the file I get following error:

Arduino: 1.8.5 (Linux), Board: "OpenCR Board, OpenCR Bootloader"

/home/das-note-2/.arduino15/packages/OpenCR/hardware/OpenCR/1.2.0/libraries/turtlebot3/examples/turtlebot3_with_open_manipulator/turtlebot3_with_open_manipulator_core/turtlebot3_with_open_manipulator_core.ino: In function 'void sendDebuglog()':
turtlebot3_with_open_manipulator_core:764: error: 'class Turtlebot3Sensor' has no member named 'getPushedBumper'
   DEBUG_SERIAL.print("Bumper : "); DEBUG_SERIAL.println(sensors.getPushedBumper());
                                                                 ^
exit status 1
'class Turtlebot3Sensor' has no member named 'getPushedBumper'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Next I commented out the following line:
           DEBUG_SERIAL.print("Bumper : "); DEBUG_SERIAL.println(sensors.getPushedBumper());
and tried to upload again, and managed to upload to the board.

However, the sample program to bringup the turtlebot3 with the open manipulator did not run, and I decided to rewrite the turtlebot3_core to the board to reset to working settings.

Now I am facing the following error, regardless of OpenCR version and what kind of sample I try to upload to the board, the IDE and the shell show:
            cmd_read_board_name fail : 0xF020

I have to note, that there is no a time value next 0xF020 comparing to https://github.com/ROBOTIS-GIT/OpenCR/issues/65

I have checked every connections several times. I can see on the board only the USB and POWER leds are on. 
I also tried to to burn the Bootloader and dowload the bootloader, also checked the USB port settings. and followed the wiki provided.
I have tried following OpenCR versions - 1.1.12, 1.1.13, 1.1.15, 1.1.0, 1.2.0 
I would like to know whether I am having problems with the hardware or is there smth wrong with my software settings?

Do you have any idea to solve this problem?
Any kind of help or advice will be appreciated!
Thanks in advance,
Husni.

routiful, 2018-06-25:
Hello @husikl :)

Sorry for inconvenience.
You can change function name from `sensors.getPushedBumper()` to `sensors.checkPushBumper()` in OpenCR versions 1.2.0.
I will be modified that as soon as possible.

Thanks

husikl, 2018-06-28:
Hello again.
I have followed your advices and understood how to work around the 
`cmd_read_board_name fail : 0xF020`.

Also by editing the arduino code to `sensors.checkPushBumper()` I can compile and upload the turtlebot3_with_open_manipulator_core.ino to the board.
However, the problem I am facing right now is, when I try to 
roslaunch turtlebot3_bringup turtlebot3_robot.launch 
There is no connection established between the board and SMC

~ roslaunch turtlebot3_bringup turtlebot3_core.launch                     
... logging to /home/das-note-2/.ros/log/1a201e5e-7a9e-11e8-8c78-9c305b264e65/roslaunch-dasnote2-5611.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:45291/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)

auto-starting new master
process[master]: started with pid [5621]
ROS_MASTER_URI=http://localhost:11311/

setting /run_id to 1a201e5e-7a9e-11e8-8c78-9c305b264e65
process[rosout-1]: started with pid [5634]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [5639]
[INFO] [1530168010.387487]: ROS Serial Python Node
[INFO] [1530168010.399627]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1530168037.861336]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

Also if I try to check only serial connectio I get the same error:
~ rosrun rosserial_python serial_node.py _baud:=115200 _port:=/dev/ttyACM0
[INFO] [1530167468.474873]: ROS Serial Python Node
[INFO] [1530167468.485684]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1530167495.908387]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

I followed this answer https://answers.ros.org/question/235188/unable-to-sync-with-device/ 
and tried to change the baud rate to 57600 and 1MHz, but I have the same results.

When I upload the turtlebot_core.ino (for turtlebot3 waffle) the serial connection is established and works fine. Also I have managed to test the open_manipulator using Dynamixel Workbench -> when I upload the usb_to_dxl.ino the connection is also established.

From these results, I would like to ask can you confirm the code in turtlebot3_with_openmanipulator_core.ino for serial connection ? Or is there a version of the code for older OpenCR version that currently works and available?

routiful, 2018-06-29:
I modified a code and tested it. You can download it below command.
Please let me know how it works

Sorry for inconvenience.

```
$ export OPENCR_PORT=/dev/ttyACM0
$ export OPENCR_MODEL=burger
$ rm -rf ./opencr_update.tar.bz2
$ wget https://github.com/ROBOTIS-GIT/OpenCR/raw/develop/arduino/opencr_release/shell_update/opencr_update.tar.bz2 && tar -xvf opencr_update.tar.bz2 && cd ./opencr_update && ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr && cd ..
```

routiful, 2018-07-02:
Hi :)

Before you run this, you are supposed to set ID to Dynamixel. Left wheel on TB3 is `1`, right wheel on TB3 is `2`, joint on manipulator is `3`~`6` and gripper is `7`.

https://github.com/ROBOTIS-GIT/OpenCR/blob/develop/arduino/opencr_arduino/opencr/libraries/turtlebot3/include/open_manipulator/open_manipulator_motor_driver.h#L33

Futhermore, you should set voltage to all Dynamixel before you run it.

We are preparing e-Manual for customer. You can see it near future.
Thanks

husikl, 2018-07-06:
Thank you for your reply!
I have set the ID to Dynamixel from 3-6 as you recommended. 
Now I can control the open_manipulator using roslaunch open_manipulator_dynamixel_ctrl dynamixel_controller.launch

However, I still don't understand how to bringup the turtlebot3 with open_manipulator together.
In the Emanual you have provided http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#hardware-setup, section 12.5  you mention 2 steps to bringup the turtlebot3 with arm:
1. Bringup TurtleBot3 with OpenManipulator. 
      by launching turtlebot3_robot.launch
2. Launch open_manipulator_with_tb3_model file. 
      by launching open_manipulator_with_tb3_model.launch use_gazebo:=false

When I connect the battery to opencr, and follow the steps you have mentioned, there is torque in wheels but no torque on the open_manipulator. I guess there is no node that starts the motors for the arm.

Also the launch files do not contain any nodes referred to open_manipulator. Is that correct? 
I have merged the most recent changes from git repo for turtlebot3.

I would like to know which nodes are you using to bringup the TurtleBot3 with OpenManipulator?
And should I add some changes to launch files mentioned above?

routiful, 2018-07-08:
Hi :)

`turtlebot3_with_open_manipulator.core` includes DynamixelSDK to control wheel and joints. 
So, you don't need to additional nodes to control Dynamixels. You can control it just send topics through `/open_manipulator_with_tb3/goal_joint_position`.

If you launch properly, the Dynamixels are moved specific position after you upload core. If not, please check ID and Baudrate(1M)

robotpilot, 2018-07-15:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-07-15 by robotpilot

erikgroving, 2018-07-18:
Hi, I have a similar issue. I commented out the DEBUG_SERIAL.print("Bumper : ") line in the arduino code to allow it to compile and then uploaded it to the OpenCR board. However, the board is no longer recognized  on any serial ports. I get an unrecognized USB device when I plug it into windows and linux just doesn't pick up on anything.

anwaybose, 2018-12-19:
Hello,
I'm also facing a problem with uploading code to OpenCR.
After performing _lsusb_ I found that my board is not in DFU mode.

output is 
_anway@simpl:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 038: ID 0483:5740 STMicroelectronics STM32F407
Bus 001 Device 003: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 002: ID 046d:c31d Logitech, Inc. Media Keyboard K200
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub_

So I have followed the steps in settiup the OpenCR and pusging the boot button and reset button simultaneously I was able to enter in DFU mode.

output is
_lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 040: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
Bus 001 Device 003: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 002: ID 046d:c31d Logitech, Inc. Media Keyboard K200
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub_

But now in DFU mode in Arduino IDE _tools -> port   /dev/ttyACM0_ is not showing

So I tried to upload the firmaware using the Shell Script but I was unable to upload.

I'm attaching the screenshot here.
[image]

Then I tried to give permission by _chmod 777 /dev/ttyACM0_
But it is also giving an error like
_chmod: cannot access '/dev/ttyACM0': No such file or directory_

_But _dmesg_  is showing the device in ttyACM0 
[75702.066990] cdc_acm 1-7:1.0: ttyACM0: USB ACM device
[75862.664973] usb 1-7: USB disconnect, device number 38
[75864.018635] usb 1-7: new full-speed USB device number 39 using xhci_hcd
[75864.168401] usb 1-7: New USB device found, idVendor=0483, idProduct=5740
[75864.168407] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[75864.168412] usb 1-7: Product: OpenCR Virtual ComPort in FS Mode
[75864.168415] usb 1-7: Manufacturer: ROBOTIS
[75864.168419] usb 1-7: SerialNumber: FFFFFFFEFFFF_

I want to know whether this a hardware problem or not?? Do I need to change my OpenCR??

OpusK, 2018-12-19:
Hi, @anwaybose 

You must use Firmware Recovery Mode, not DFU mode for downloading firmware.
Please refer to below link.

 - [USB port setting](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#usb-port-settings)
 - [Firmware Recovery Mode](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#firmware-recovery-mode)

anwaybose, 2018-12-19:
Hi, @OpusK 
Thanks for your suggestion.
I have tried accordingly following the link, but still it is of no use

_anway@simpl:~$ sudo opencr_ld /dev/ttyACM0 115200 ./opencrfw.bin 1
sudo: opencr_ld: command not found_

OpusK, 2018-12-19:
@anwaybose,

anwaybose, 2018-12-19:
@OpusK
I tried to eport binary sketch from Arduino IDE after entering into the Firmware Recovery Mode, but it is throwing error

_Board OpenCR (platform OpenCR, package OpenCR) is unknown

Error compiling for board OpenCR Board._

anwaybose, 2018-12-19:
@OpusK 
After reinstalling the driver and burning the bootloader, I'm still unable to upload my code. When I tried to do so, it is thrwing an error as follow:
[image]
Please confirm whether this is an hardware issue or not??

anwaybose, 2018-12-20:
@OpusK ,
 I have tried this
> > anway@simpl:~$ sudo opencr_ld /dev/ttyACM0 115200 ./opencrfw.bin 1
> 
> This command should be run from where the opencr_ld and opencrfw.bin files are located.

But it is also throwing some error as follow:

_anway@simpl:~/Dev/OpenCR/arduino/opencr_arduino/tools/opencr_tools_1.0.0/linux64$ ./opencr_ld /dev/ttyACM0 115200 ./turtlebot3_core.ino.OpenCR.bin 1
opencr_ld ver 1.0.4
opencr_ld_main 
>>
file name : ./turtlebot3_core.ino.OpenCR.bin 
file size : 177 KB
Open port OK
Clear Buffer Start
Clear Buffer End
cmd_read_board_name fail : 0xF020_

OpusK, 2018-12-20:
@anwaybose ,

> When the boot loader is running, the STATUS LED blinks every 100ms.

This error occurs because the board name can not be read normally. As I mentioned above, when the OpenCR enters Firmware recovery mode, does the STATUS LED blink?

anwaybose, 2018-12-20:
@OpusK 
Yes it blinks at a frequency of 100ms

anwaybose, 2018-12-20:
@OpusK,
I am not using any usb hub. I have also checked with changing the usb cable, the problem persists. My usb cables are working fine I have checked them with other devices. Can it be a problem with the micro USB pin attached to the OpenCR?

OpusK, 2018-12-21:
@anwaybose 

> Can it be a problem with the micro USB pin attached to the OpenCR?

You said you uploaded the bootloader normally using DFU(USB)?
If so, this is not the problem you mentioned.

Would you like to check that the generic blink example is uploaded instead of turtlebot3_core.ino?

anwaybose, 2018-12-21:
@OpusK 
> Would you like to check that the generic blink example is uploaded instead of turtlebot3_core.ino?

I have tried to upload it, but in tools -> port nothing is showing, so it failed to upload the code from Arduino IDE. 
But I have tried in this way:
_anway@simpl:~/Dev/OpenCR/arduino/opencr_arduino/tools/opencr_tools_1.0.0/linux64$ ./opencr_ld /dev/ttyACM0 115200 ./Blink.ino.OpenCR.bin 1
opencr_ld ver 1.0.4
opencr_ld_main 
>>
file name : ./Blink.ino.OpenCR.bin 
file size : 37 KB
Open port OK
Clear Buffer Start
Clear Buffer End
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 0.997000 sec
flash_write : 0 : 0.861000 sec 
CRC OK 38E4F0 38E4F0 0.001000 sec
[OK] Download 
jump_to_fw 
jump finished_

And I'm able to upload the code.

Then waht is the error you think?

OpusK, 2018-12-21:
> I have tried to upload it, but in tools -> port nothing is showing, so it failed to upload the code from Arduino IDE.

Since you have successfully uploaded via the command, this is likely a problem with the Arduino IDE.
Have you tried uploading the turtlebot_core even by command method?

anwaybose, 2018-12-21:
@OpusK,
Yes today I'm able to upload the turtlebot3_core by command method. So what do you suggest? Do I need to reinstall my Arduino IDE?

OpusK, 2018-12-21:
@anwaybose,

Arduino simply executes the commands you have executed manually for you.
So, I do not know the exact cause...
I suggest you try reinstalling your Arduino IDE to the latest version.

anwaybose, 2018-12-21:
@OpusK ,
Thanks for your suggession.

OpusK, 2018-12-21:
@anwaybose, 

Because it works with the command, it seems difficult to specify the cause. I have tested on various PCs in the company, but unfortunately the same symptoms are not reproducible and I can not debug ...

anwaybose, 2018-12-21:
@OpusK 
Thank You.
After re-installing Arduino IDE it is working properly. But now I'm facing a new problem. I have opened a new issue  #361 . Can you help me in resolving this.

ProfJust, 2021-01-13:
I had the same Error with TurtleBot3 and the openCR-Board.

 >cmd_read_board_name fail : 0xF020

Change USB-PowerCable to an USB-Data and Power-Cable then it works!
With the correct cable, you can use the /dev/ttyACM0-Port.
With the wrong cable you can only see /dev/ttyS4 in the Arduino-IDE and this wasn't working.