ROBOTIS-GIT/turtlebot3 #241: Questions about base mesh files

Affonso-Gui, 2018-06-26:
Hello all,

I am doing some modifications on the Turtlebot3 Waffle frames to have space for some more sensors, as you can see below. I am working at a forked workspace on Onshape, and exporting it to use as a base mesh in the urdf.

[image]

So here are my questions:
-  Why are the meshes in this directory (e.g. `waffle_base.stl`) different from the ones exportable by the Onshape workspace (plates have less holes on it, etc)? Is it to make the files smaller? If so can we have access to editables for such plates?
- Is there any reasons for the files being .stl rather than .dae? When exporting from Onshape collada files appear to be smaller on size, preserve color and do not need the `scale` or `material` tags in the urdf.

routiful, 2018-06-27:
Hi @Affonso-Gui :)

Thank you for your inquiry about TurtleBot3!
The origin of waffle_base has too many surface to load on Rviz or Gazebo even size is too big. So we uploaded modified waffle_base to Github. Unfortunately, we don't have provided editable version of it.

We already know .dae file is recommended by OSRF. We have a plan to support it near future.

Thanks

Affonso-Gui, 2018-06-27:
Hello @routiful,

Thank you for clarifying. Looking forward for .dae support.

closed 2018-06-27 by Affonso-Gui

routiful, 2018-06-27:
Nice. If you need any help, please feel free to open issue :)