ROBOTIS-GIT/turtlebot3 #242: turtlebot3 bringup incomplete tf

tedalias, 2018-06-26:
1. Which TurtleBot3 you have?

    - [ ] Burger
    - [x] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [x] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [x] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)
 
4. Write down the commands you used in order

    [Remote PC] roscore
    [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
    [Remote PC] rqt
 
6. Please, describe detailedly what difficulty you are in 

Hi,

I'm relatively new to ROS and the turtlebot.

When I try to bringup the turtlebot, there is no error message, but the tf tree seems to be incomplete.

The only transformations broadcasted are odom -> base_footprint and base_link -> imu_link.

So when I start rviz for example it misses some parts of tf (eg for the laser scanner).

How can I bring up the remaining transformations? Where are they called from? As far as I can tell, in the Arduino file, only the above mentioned transformations are broadcasted.

Thanks in advance!

robotpilot, 2018-06-26:
Hi @tedalias,

When you run the launch file below on your Remote PC, you will get information about the robot model. You will then be able to see all of the tf information you want.

```
roslaunch turtlebot3_bringup turtlebot3_remote.launch
```

https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_bringup/launch/turtlebot3_remote.launch

Please refer to the following manual for more details on bringup.

http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup-a-turtlebot3-burger

tedalias, 2018-06-27:
Thanks, @robotpilot , that was the part I was missing. It works now.
Closing the issue

closed 2018-06-27 by tedalias