ROBOTIS-GIT/turtlebot3 #244: Turtlebot3 Waffle Pi not Publishing/Listening to topics

erikgroving, 2018-07-03:
ISSUE TEMPLATE ver. 0.1.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [X ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [X ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [X ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

    - Software version: [1.2.0]
    - Firmware version: [1.2.0]
 
4. Write down the commands you used in order

    [Remote PC] roscore
    [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
    [Remote PC] roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
 
5. Copy and Paste your error message on terminal

No error messages are occurring
  
6. Please, describe detailedly what difficulty you are in 

    I have brought up the turtlebot3 with both the ubuntu mate and raspbian systems with the commands specified in step 4 in the exact order. No error messages are shown, however, the turtlebot is not publishing or subscribing to any of the topics. The topics are indeed created though, as they are visible on the topic monitor. If I manually publish to a topic that the turtlebot is subscribed to, nothing happens, though the topic will update with the new values on the topic monitor. I am not sure what to do to make the turtlebot waffle pi listen to its subscribed topics and actively publish to its topics. Any ideas?

kijongGil, 2018-07-04:
Hi @erikgroving :)
Did you set up turtlebot3's IP and remote PC's IP?
http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#5-network-configuration

If you want to check publisher and subscriber node of cmd_vel, enter the following command.

`$ rostopic info /cmd_vel`

I think you have a publisher node, no subsrciber.

erikgroving, 2018-07-04:
Hi, this is the output of the command:

```
Type: geometry_msgs/Twist

Publishers: 
 * /turtlebot3_teleop_keyboard (http://140.113.216.14:33027/)

Subscribers: 
 * /turtlebot3_core (http://192.168.29.247:40128/)
```

We can see that the subscriber is the turtlebot3_core and the publisher is the remote pc, however, no reaction is occurring from the turtlebot. Likewise, the turtlebot isn't publishing to any topics. The output of `rostopic info /battery_info` is:

```
Type: sensor_msgs/BatteryState

Publishers: 
 * /turtlebot3_core (http://192.168.29.247:40128/)

Subscribers: 
 * /rqt_gui_py_node_6881 (http://140.113.216.14:37384/)

```

kijongGil, 2018-07-04:
Did you connect the remote PC and the robot to the same internet?

erikgroving, 2018-07-04:
That was the issue. I was connecting to the server in our lab that was not on the same router, though the server could still be found by the raspberry pi. Upon using an ethernet cable from the router for the remote pc, the raspberry pi was able to publish live data. All solved! Thanks.

closed 2018-07-04 by erikgroving

kijongGil, 2018-07-04:
I am glad that the problem has been solved. :)
If you have another problem, please reopen it.
Thanks.