ROBOTIS-GIT/turtlebot3 #246: Turtlebot 3 Rviz No transform from [wheel_left_link] to [base_footprint]

shrivats1995, 2018-07-05:
ISSUE TEMPLATE ver. 0.1.0

1. Which TurtleBot3 you have?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [X] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

    - Software version: [1.12.13]
    - Firmware version: [1.2.0]
 
4. Write down the commands you used in order

    [Remote PC] roscore
    [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
    [Remote PC] $ export TURTLEBOT3_MODEL=burger
    [Remote PC] $ roslaunch turtlebot3_bringup turtlebot3_remote.launch
    [Remote PC]$ rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz
 
5. Copy and Paste your error message on terminal

[image]

6. Please, describe detailedly what difficulty you are in 

Hi,
I am using a turtlebot 3 burger with ubuntu mate installed on the raspberry pi and the remote PC is running Ubuntu  16.04 ( Xenial Xerus )
I was working on the 'bringup' section of the tutorial and while running rviz, I got the following status error in the robot model section.
No transform from [wheel_left_link] to [base_footprint]
No transform from [wheel_right_link] to [base_footprint]
I have updated the firmware on the openCR board to the latest version (1.2.0)

kijongGil, 2018-07-06:
Hi :) @shrivats1995
Make sure your RemotePC and turtlebot3's IP.
Follow link will help you.
https://github.com/ROBOTIS-GIT/turtlebot3/issues/229

Thanks,
Gilbert.

shrivats1995, 2018-07-06:
Hi Gilbert,

Thanks a lot! Will check that.

Regards,
Shrivatsan

shrivats1995, 2018-07-06:
Changing the IP worked. Thank you so much for the help!
I'll close the issue.

closed 2018-07-06 by shrivats1995

kijongGil, 2018-07-07:
I am glad that the problem has been solved. :)
If you have another problem, please reopen it.

dhikihandika, 2019-04-27:
hi gilbert,

My robot have same problem it. I have done check connection RemotePC and TurtleBot3 IP, this is fine.
But my robot still eror in Rviz;
No transform from [wheel_left_link] to [base_footprint], and
No transform from [wheel_right_link] to [base_footprint]

Please show me instruction to fix it !!!
Thank you,

valazio195, 2019-06-12:
Halo bro dhikiandika.

Kalau emang dari setup IP address, konfigurasi OpenCR udah oke2 aja, tapi pas ngetes kendala rviz-nya masih ada (wheel right and left missing di Robot Model), coba lo konfigurasi ulang IP address both on Remote PC and Turtlebot.

Karena pada dasarnya kita mau robotnya autonomous, maka Turtlebot-nya harus host ROS MASTER, maka both ROS_MASTER_URI dan ROS_HOSTNAME_nya harus diisi ama IP address dari Turtlebot itu sendiri.

Untuk remote PC-nya, ganti IP Address dari ROS_MASTER_URI ke IP turtlebot lo, nah kalau IP dari ROS_HOSTNAME-nya yah IP address dari komputer lo

TurtleBot's netbook hosts the ROS Master with
ROS_MASTER_URI= http://<IP address of TurtleBot>:11311
and
ROS_IP=http://<IP address of TurtleBot>
•	 The remote computer has
ROS_MASTER_URI = http://<IP address of TurtleBot>:11311
and
ROS_IP==http://<IP address of remote computer>

jangan lupa edit2 itu paka nano, trus kalau udah, langsung source

valazio195, 2019-06-13:
Sorry2 bro dhikiandika, looks like you don't need to configure your IP or your open CR again.

If your problem still persist nevertheless, maybe it's because your firewall still turned on in your laptop.

Try sudo ufw disable command first, see what happen next

dhikihandika, 2019-06-13:
oke bro..
I will try your solution next few days..
Thanks for his help and i hope your solution make completed my problem.

LouisValery, 2020-01-15:
If you used this node:
<node pkg="tf" type="static_transform_publisher" name="odom_1_map_broadcaster" args="0 0 0 0 0 0 map robot1_tf/odom 100"/>
make sure that your map is linked to "odom" and not "footprint".
It solved my problem.