ROBOTIS-GIT/turtlebot3 #249: TurtleBot3 Bringup ROS Unable to sync issue

Hellsgone, 2018-07-10:
ISSUE TEMPLATE ver. 0.1.0

1. Which TurtleBot3 you have?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ x] Raspbian (with pre installed ros packages as in told here http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#raspberry-pi-3-setup

    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [x ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down the commands you used in order

    [Remote PC] roscore
    [SBC] sudo apt-get install ros-kinetic-rosserial-python ros-kinetic-tf
    [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
 
6. Copy and Paste your error message on terminal
  
 -------------------------------------------------[REMOTE PC]---------------------------------------------
ardadenker@ardadenker-VirtualBox:~$ roscore
... logging to /home/ardadenker/.ros/log/1f626a1e-8428-11e8-869c-0800275e3e55/roslaunch-ardadenker-VirtualBox-6237.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.1.33.103:37601/
ros_comm version 1.12.13

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES

auto-starting new master
process[master]: started with pid [6249]
ROS_MASTER_URI=http://10.1.33.103:11311/

setting /run_id to 1f626a1e-8428-11e8-869c-0800275e3e55
process[rosout-1]: started with pid [6262]
started core service [/rosout]

------------------------------------[TURTLEBOT3 PC (Raspberry Pi 3)] -----------------------------------------

pi@raspberrypi:~ $ sudo apt-get install ros-kinetic-rosserial-python ros-kinetic-tf
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-rosserial-python
E: Unable to locate package ros-kinetic-tf
pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/pi/.ros/log/1f626a1e-8428-11e8-869c-0800275e3e55/roslaunch-raspberrypi-3472.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.1.83.65:46405/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://10.1.33.103:11311

process[turtlebot3_core-1]: started with pid [3481]
process[turtlebot3_lds-2]: started with pid [3482]
process[turtlebot3_diagnostics-3]: started with pid [3483]
[INFO] [1531221031.731335]: ROS Serial Python Node
[INFO] [1531221031.777864]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1531221048.895325]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
^C[turtlebot3_diagnostics-3] killing on exit
[turtlebot3_lds-2] killing on exit
[turtlebot3_core-1] killing on exit
[INFO] [1531221052.823853]: Send tx stop request
shutting down processing monitor...
... shutting down processing monitor complete
done
pi@raspberrypi:~ $ 
   ```
  
7. Please, describe detailedly what difficulty you are in 

    - I have followed the emanual 6. Setup guide and done every step however I came across with this issue when I tried to use SLAM which in order to do that I need to use roscore is in found here http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#run-roscore but I am unable to connect. roscore is used in the remote pc, bashrc is setup correctly as well as on turtlebot pc. roslaunch is used in turtlebot pc which is where the issue comes. Raspberry pi is both connected to opencr and slam, hardware setup is done correctly as well. ssh works too. I have not used the commands on 9. SLAM, I am trying to use 7. Bringup 
As I am very new to ubuntu, raspberry pi and turtlebot itself it will be much appreciated if I could get some help here. Thank you!

![er](https://user-images.githubusercontent.com/41053677/42508358-250651f2-8440-11e8-9926-b393aea21458.png)

Hellsgone, 2018-07-10:
Quick update, after following 9. Slam guide and using following code 
roslaunch turtlebot3_bringup turtlebot3_lidar.launch   [TurtleBot PC]
roslaunch turtlebot3_bringup turtlebot3_model.launch  [Remote PC]
roscore  [Remote PC]
 RViz launches and I can see the point clouds. However roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch command does not work for some reason. There is no error and I have uploaded the turtlebot3_core program to opencr, the dynamixels light up when powered. 

Using roslaunch turtlebot3_bringup turtlebot3_robot.launch on Turtlebot PC now gives me this error shown below..
                       .I
                       .I
                      V

pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/pi/.ros/log/f5b0d536-8452-11e8-869c-0800275e3e55/roslaunch-raspberrypi-1137.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.1.83.65:35411/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://10.1.33.103:11311

process[turtlebot3_core-1]: started with pid [1146]
process[turtlebot3_lds-2]: started with pid [1147]
process[turtlebot3_diagnostics-3]: started with pid [1148]
[INFO] [1531236201.966891]: ROS Serial Python Node
[INFO] [1531236202.027224]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1531236204.266476]: Tried to publish before configured, topic id 133
[ERROR] [1531236204.292876]: Tried to publish before configured, topic id 127
[ERROR] [1531236204.299015]: Tried to publish before configured, topic id 127
[INFO] [1531236204.362469]: Note: publish buffer size is 1024 bytes
[INFO] [1531236204.364878]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1531236204.399880]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1531236204.414143]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1531236204.434025]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1531236207.836800]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1531236207.859107]: Note: subscribe buffer size is 1024 bytes
[INFO] [1531236207.860967]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1531236207.890601]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1531236207.921674]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1531236207.946674]: Setup subscriber on reset [std_msgs/Empty]
[ERROR] [1531236207.953226]: Tried to publish before configured, topic id 127
[ERROR] [1531236208.012403]: Tried to publish before configured, topic id 127
[ERROR] [1531236208.032885]: Tried to publish before configured, topic id 128
[ERROR] [1531236208.039628]: Tried to publish before configured, topic id 127
[INFO] [1531236208.052787]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1531236208.060577]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1531236208.064972]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1531236208.069514]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1531236208.084457]: Tried to publish before configured, topic id 128
[ERROR] [1531236208.091604]: Tried to publish before configured, topic id 128
[ERROR] [1531236208.164430]: Tried to publish before configured, topic id 127
[ERROR] [1531236208.211198]: Tried to publish before configured, topic id 127
[ERROR] [1531236208.282778]: Tried to publish before configured, topic id 127
[ERROR] [1531236208.290668]: Tried to publish before configured, topic id 128
[ERROR] [1531236208.339135]: Tried to publish before configured, topic id 127
[ERROR] [1531236208.345164]: Tried to publish before configured, topic id 127
[ERROR] [1531236208.356841]: Tried to publish before configured, topic id 127
[INFO] [1531236208.375337]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1531236208.381993]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1531236208.388273]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1531236208.393231]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1531236208.404847]: Tried to publish before configured, topic id 128
[ERROR] [1531236208.411283]: Tried to publish before configured, topic id 128
[ERROR] [1531236208.420272]: Creation of publisher failed: Checksum does not match: d537ed7b8d95065b6c83830430b93911,7250c1dc0b61c4190e78f528f599285f
[INFO] [1531236209.192650]: Setup publisher on version_info [turtlebot3_msgs/VersionInfo]
[turtlebot3_diagnostics-3] killing on exit
[INFO] [1531236209.198669]: Setup publisher on imu [sensor_msgs/Imu]
[turtlebot3_lds-2] killing on exit
[turtlebot3_core-1] killing on exit
[INFO] [1531236209.204748]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1531236209.212129]: Send tx stop request
shutting down processing monitor...
... shutting down processing monitor complete
done
pi@raspberrypi:~ $ 

Currently I am trying to update to ROS 2 from https://discourse.ros.org/t/turtlebot3-software-and-firmware-update-and-waffle-pi/3729

Hellsgone, 2018-07-10:
I have realized that [ roslaunch turtlebot3_bringup turtlebot3_robot.launch ] should be used in Remote PC as well, that way teleop works but with only left dynamixel moving. I have tried the slam nodes and teleportation nodes on turtlebot but it gave me this error (turtlebot3_slam.launch is neither a launch file in a package nor is a launch file name). What I don't understand is everything works only when opencr is connected to Remote PC and not TurtleBot PC. To tidy up all this mess, I would conclude it here

-Remote PC-
1. roscore
2. roslaunch turtlebot3_bringup turtlebot3_robot.launch
3. roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

-TurtleBot PC-
1. roslaunch turtlebot3_bringup turtlebot3_lidar.launch

-Remote PC-
 3. roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=cartographer (or any other slam node)

This way I can see it's working with any slam node, as a proof I have attached the picture.
[image]

Could you assist me if anything I am doing is wrong?
Thank you.

kijongGil, 2018-07-11:
Hi @Hellsgone :)
Please update your OpenCR. Maybe your problem will be solved.
http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-firmware-upload-for-tb3

Thanks,
Gilbert

Hellsgone, 2018-07-11:
Hi, thank you for the reply. First I have tried ssh and followed the commands in the link you sent. While this was happening, raspberry pi 3 was connected to both LD sensor and opencr. The following code went on successfully. After that I tried bringup and robot, still in ssh, but I got the same error. 

[Remote PC] using ssh and updating OpenCR

ardadenker@ardadenker-VirtualBox:~$ ssh pi@10.1.83.65
pi@10.1.83.65's password: 
Linux raspberrypi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Jul 10 15:39:59 2018 from 10.1.33.103
pi@raspberrypi:~ $ export OPENCR_PORT=/dev/ttyACM0
pi@raspberrypi:~ $ export OPENCR_MODEL=burger
pi@raspberrypi:~ $ rm -rf ./opencr_update.tar.bz2
pi@raspberrypi:~ $ wget https://github.com/ROBOTIS-GIT/OpenCR/raw/master/arduino/opencr_release/shell_update/opencr_update.tar.bz2 && tar -xvf opencr_update.tar.bz2 && cd ./opencr_update && ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr && cd ..
--2018-07-11 09:39:24--  https://github.com/ROBOTIS-GIT/OpenCR/raw/master/arduino/opencr_release/shell_update/opencr_update.tar.bz2
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/shell_update/opencr_update.tar.bz2 [following]
--2018-07-11 09:39:25--  https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/shell_update/opencr_update.tar.bz2
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.0.133, 151.101.64.133, 151.101.128.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.0.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 304853 (298K) [application/octet-stream]
Saving to: 'opencr_update.tar.bz2'

opencr_update.tar.b 100%[===================>] 297.71K  1.77MB/s    in 0.2s    

2018-07-11 09:39:29 (1.77 MB/s) - 'opencr_update.tar.bz2' saved [304853/304853]

opencr_update/burger_turtlebot3_core.ino.bin
opencr_update/released_1.2.0.txt
opencr_update/waffle_turtlebot3_core.ino.bin
opencr_update/opencr_ld_shell_arm
opencr_update/update.sh
opencr_update/
opencr_update/waffle.opencr
opencr_update/opencr_ld_shell_x86
opencr_update/burger.opencr
armv7l
arm
OpenCR Update Start..
opencr_ld_shell ver 1.0.0
opencr_ld_main 
[  ] file name   	: burger.opencr 
[  ] file size   	: 176 KB
[  ] fw_name     	: burger 
[  ] fw_ver      	: 1.2.0 
[OK] Open port   	: /dev/ttyACM0
[  ]
[  ] Board Name  	: OpenCR R1.0
[  ] Board Ver   	: 0x17020800
[  ] Board Rev   	: 0x00000000
[OK] flash_erase 	: 1.00s
[OK] flash_write 	: 1.98s 
[OK] CRC Check   	: 11FB73C 11FB73C , 0.006000 sec
[OK] Download 
[OK] jump_to_fw 
pi@raspberrypi:~ $ 
I
I
I
[Remote PC] using ssh
pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/pi/.ros/log/6ce36522-84e6-11e8-8d32-0800275e3e55/roslaunch-raspberrypi-1333.log
[... 141 lines cut ...]
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://10.1.25.91:11311

process[turtlebot3_core-1]: started with pid [6393]
process[turtlebot3_lds-2]: started with pid [6394]
process[turtlebot3_diagnostics-3]: started with pid [6395]
[ERROR] [1531299591.800440893]: An exception was thrown: open: No such file or directory
[turtlebot3_lds-2] process has died [pid 6394, exit code 255, cmd /opt/ros/kinetic/lib/hls_lfcd_lds_driver/hlds_laser_publisher __name:=turtlebot3_lds __log:=/home/ardadenker/.ros/log/9dd02bfa-84e8-11e8-8d32-0800275e3e55/turtlebot3_lds-2.log].
log file: /home/ardadenker/.ros/log/9dd02bfa-84e8-11e8-8d32-0800275e3e55/turtlebot3_lds-2*.log
[INFO] [1531299592.166426]: ROS Serial Python Node
[INFO] [1531299592.175610]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1531299592.178153]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
[turtlebot3_core-1] process has finished cleanly
log file: /home/ardadenker/.ros/log/9dd02bfa-84e8-11e8-8d32-0800275e3e55/turtlebot3_core-1*.log
I
----------------------------------------------Conclusion------------------------------------------------------
Everything only works when opencr is connected to remote pc where it can follow the commands for slam nodes. However, my goal is to use raspberry pi 3 for that. At this point the microUSB cable for SBC to OpenCR is useless. Any suggestions?
Thank you :)

Hellsgone, 2018-07-11:
@kijongGil I realized that while [ roslaunch turtlebot3_bringup turtlebot3_robot.launch ] is being executed in the Remote PC, I see these error comes up however after that everything still works in the setup I described before. Could this be the root of the problem? 

ardadenker@ardadenker-VirtualBox:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/ardadenker/.ros/log/9dd02bfa-84e8-11e8-8d32-0800275e3e55/roslaunch-ardadenker-VirtualBox-7427.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.1.25.91:44201/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://10.1.25.91:11311

process[turtlebot3_core-1]: started with pid [7444]
process[turtlebot3_lds-2]: started with pid [7445]
process[turtlebot3_diagnostics-3]: started with pid [7458]
**[ERROR] [1531301078.765680162]: An exception was thrown: open: No such file or directory
[turtlebot3_lds-2] process has died [pid 7445, exit code 255, cmd /opt/ros/kinetic/lib/hls_lfcd_lds_driver/hlds_laser_publisher __name:=turtlebot3_lds __log:=/home/ardadenker/.ros/log/9dd02bfa-84e8-11e8-8d32-0800275e3e55/turtlebot3_lds-2.log].**
_log file: /home/ardadenker/.ros/log/9dd02bfa-84e8-11e8-8d32-0800275e3e55/turtlebot3_lds-2*.log_
[INFO] [1531301079.238646]: ROS Serial Python Node
[INFO] [1531301079.252550]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1531301081.401617]: Note: publish buffer size is 1024 bytes
[INFO] [1531301081.403701]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1531301081.420916]: Setup publisher on version_info [turtlebot3_msgs/VersionInfo]
[INFO] [1531301081.460832]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1531301081.465278]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1531301081.474091]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1531301081.481837]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1531301081.486142]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1531301081.490416]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1531301082.043251]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1531301082.048211]: Note: subscribe buffer size is 1024 bytes
[INFO] [1531301082.048640]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1531301082.053794]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1531301082.060660]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1531301082.065057]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1531301084.482310]: --------------------------
[INFO] [1531301084.485167]: Connected to OpenCR board!
[INFO] [1531301084.487166]: This core(v1.2.0) is compatible with TB3 Burger
[INFO] [1531301084.489918]: --------------------------
[INFO] [1531301084.491918]: Start Calibration of Gyro
[INFO] [1531301087.018977]: Calibration End

Hellsgone, 2018-07-11:
Quick Update: I have tried using ssh into sbc and executed these commands. This time, OpenCR is connected to SBC
$ roslaunch turtlebot3_bringup turtlebot3_lidar.launch
$ roslaunch turtlebot3_bringup turtlebot3_core.launch
Now these work instead of the turtlebot3_robot but again the turtlebot3_core.launch gives an error while it executes but functions properly after finishes.
**[ERROR] [1531303834.074751]: Creation of publisher failed: Checksum does not match: d537ed7b8d95065b6c83830430b93911,7250c1dc0b61c4190e78f528f599285f**
After the bringup, rViz successfully shows the point clouds in remote pc (7. Bringup) however rqt does not show anything, topic monitor launches but it's blank and also teleop or any other command that moves the robot did not work. Dynamixels did not work when using core and lidar on SBC while it works using robot.launch in Remote PC. One other issue is that only the left dynamixel is working, I tried only plugging in one or same port as left one was working but couldnt get right one to work.

kijongGil, 2018-07-12:
Hi Hellsgone,
I'm sorry, you have too many questions and it is hard for me to understand.
Let's solve the problem step by step. First, Check your turtlebot3 packages and OpenCR.
Remove turtlebot3 packages and reinstall it (turtlebot3, turtlebot3_msgs, etc) and update OpenCR again.
Then, run roscore in Remote PC.
`$ roscore`

Run turtlebot3_bringup in SBC
`$ roslaunch turtlebot3_bringup turtlebot3_robot.launch`

New terminal in Remote PC
` $ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch`

Check your turtlebot3 is working and tell me.

Hellsgone, 2018-07-12:
@kijongGil 
Yes, I apologies with all the questions and things I wrote down, I am trying my best to keep my progress updated so it won't confuse you. 
So, first of all, I reinstalled the rasbian and intsalled the packages. I followed the ros 2 guide as well https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888
in turtlebot3 everything went normally and no errors. However when doing the last step on remote pc, cd ~/catkin_ws && catkin_make, this command gave me this error, **CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.** so the remote pc part is incomplete however I carried on as you told me to use those commands after reinstalling. And it works without a single error. I tried slam nodes and navigation, everything works perfectly. The robot does a 180 degree turn each time it reaches the goal position I'm not sure that it is suppose to. My last question would be if that cmake error something important, if not then you can close this issue because you solved my problem. Thanks a lot!

kijongGil, 2018-07-13:
What is mean ros 2 you said? Because ROS 2 is a new version of ROS that is under heavy development. But,  this link you said is not about ROS 2, just about updating our software.
I think your problem is not related to ROS 2. 
CMake error is that there are packages that can not be built with cmake in your workspace.
Back up or remove all package in your workspace without turtlebot3 packages, then try to build again.

kijongGil, 2018-07-13:
I am glad that the problem has been solved. :)
If you have another problem, please reopen it.
Thanks.