ROBOTIS-GIT/turtlebot3 #250: Install Kinect FOR 3D vision: Burger+Raspberry Pi 3+Ubuntu MATE 16.04

stevensu1838, 2018-07-11:
Hi team,

I have both a turtlebot2 and turtlebot3 Burger. Now I want to put the Kinect V1 sensor from Trutlebot2 on the turtlebot3 Burger to do mapping. I think I need to make some change in the code. Can you please give me some suggestions?
Thanks a lot.

kijongGil, 2018-07-12:
Hi @stevensu1838,
You have to install freenect packages for using kinect.
`$ sudo apt-get install ros-kinetic-freenect*`

and run the camera.
`$ roslaunch freenect_launch freenect.launch` 

But, I don't recommend kinect. How about use Raspberry pi camera?

stevensu1838, 2018-07-13:
Hi, thank u so much for your reply. The thing is I want to do Octomap with Kinect camera. And I've done it with Kinect on Turtlebot2. Now I wanna do the same mapping with Turtlebot3. I've got a Pi camera. Any suggestions? Can I do Octomap with the Intel realsense camera for Waffle? Thanks a lot.

kijongGil, 2018-07-16:
Sorry for the late reply.
I understand your idea. Actually, Pi camera can't make a 3D map.
And realsense camera has 3D depth data. So you can make a 3D map.

YuehChuan, 2018-07-21:
It's possible to use realsense D435 camera on raspberry pi3.
https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md
I don't have a D435 with me right now.

stevensu1838, 2018-07-21:
Hi Gilbert,
Please find the linked website I used to do Octomap with Turtlebot2, can you please kindly tell me How can I use It with Turtlebot3 with NUC and Kinect on board? Thanks a lot.

http://ros-developer.com/2017/05/02/making-occupancy-grid-map-in-ros-from-gazebo-with-octomap/

stevensu1838, 2018-07-21:
Hi all, Thanks a lot for your advice. I've got a NUC pc anD will try to put it on the robot. My purpose is to do Octomap with my Tutlebot3 with Kinect. (I've done the Octomap with Turtlebot2 successfully.)My question is do you think the Octomap map can be done with my Turtlebot3(NUC PC and Kinect sensor on board)? And Is the process (like command lines)much different from Turtlebot2? Thanks a lot

routiful, 2018-07-23:
Because the latest question is same as #254, this issue will be closed.

closed 2018-07-23 by routiful