ROBOTIS-GIT/turtlebot3 #253: Scan message publish rate of the Turtlebot3 LIDAR

SHTseng, 2018-07-15:
Hi, 
I have a Turtlebot3 waffle, which equipped with a LDS-01 LIDAR. I use the standard launch file to start the LIDAR. 
`roslaunch turtlebot3_bringup turtlebot3_lidar.launch`

However, my measuring of the /scan message publish rate is about 5Hz, and my application needs higher rate. I wonder if there is any way to increase the /scan message publish rate?

robotpilot, 2018-07-15:
Hi @SHTseng,

Unfortunately, the setting is maximized. Since it is the setting specified by the manufacturer, there is no way to customize it until you use another LiDAR.

SHTseng, 2018-07-16:
Thanks for the reply, then I think the update_rate for the gazebo laser plugin should be 5, not 1800.

robotpilot, 2018-07-17:
Hi @SHTseng,
Yes. Your point is correct. The program of the real robot is so, but our URDF for simulation is '1800' you said. A value of '1800' is the 'sampling rate', not 'update rate'. We also used this value incorrectly, as shown in the code below.
https://github.com/ROBOTIS-GIT/turtlebot3/blob/75b685241f9b7e448ecdbf107296d2de08f842e2/turtlebot3_description/urdf/turtlebot3_burger.gazebo.xacro#L125-L129
Thank you so much. We will correct this to '5'. This issue is the part you found, so if you do a Pull Request we will get your PR. Do you want a PR? Or are we going to fix it? :)

robotpilot, 2018-07-18:
Thanks PR https://github.com/ROBOTIS-GIT/turtlebot3/pull/258, I would greatly appreciate it.
I'm going to close this issue. Please feel free to reopen this thread if there's any opinion to throw.
Once again, thank you for your help and supports. :)

closed 2018-07-18 by robotpilot

LEZHOU26, 2019-09-19:
Hi, I met a problem when I install the Lidar on the turtlebot 3, when I try to run

$ sudo chmod a+rw /dev/ttyUSB0

It just said cannot access dev/ttyUSB0: no such file or directory.
Did you encounter this problem?