ROBOTIS-GIT/turtlebot3 #254: Update my Burger to Waffle, 	I need to by Intel® Joule™ 570x or I have other options?

stevensu1838, 2018-07-16:
Hi team,
I am planning to update my Burger to a Waffle. Coz I think it will be more powerful to support a 3D camera(I might be wrong). So I want to buy an Intel® Joule™ 570x, but Intel discontinued the products. My question is do I sill try to buy a Joule or I have other options as the onboard PC? What is your plan?
Secondly, I also wanna buy an Openmanipulator. Will you sell it as a whole thing SOON or I still need to 3D print the link parts. Thanks a lot.

Steven

kijongGil, 2018-07-16:
Hi @ stevensu1838 :)
You known as intel discontinued the  Joule™ 570x, So we  sell waffle_Pi with Raspberry pi.
But, you want to use 3D camera. I recommend using upboard + kinetic camera. 

Thanks for your interesting about Openmanipulator. Maybe you can buy it in September.
There are 3D print the link parts you wanted follow link.
http://emanual.robotis.com/docs/en/platform/openmanipulator/#hardware-setup

Thanks,
Gilbert

DoNck, 2018-07-20:
Lost inside Intel's sites hierachy, there is an offer for a pretty cheap kit including both the Up! board and the sensor :
https://click.intel.com/intelr-realsensetm-robotic-development-kit-2622.html
It is not the best sensor on the market but I think it can be a good start :-)

stevensu1838, 2018-07-21:
Hi Gilbert,
Please find the linked website I used to do Octomap with Turtlebot2, can you please kindly tell me How can I use It with Turtlebot3 with NUC and Kinect on board? Thanks a lot.

http://ros-developer.com/2017/05/02/making-occupancy-grid-map-in-ros-from-gazebo-with-octomap/

routiful, 2018-07-23:
Hi :)

After bringup turtlebot3 properly, you can use same launch commands when you used to do Octomap with TurtleBot2. Please try it first and ask question when you face problems.

Furthermore, OpenManipulator will be released soon. You can buy it this fall or winter.

stevensu1838, 2018-07-24:
Hi Gilbert, Thanks lot for your advice.
Right now I am following the following tutorial for 3D Visualization with turtlebot2 to visualize with my turtlebot3 and Kinect sensor(our ultimate target is to do Octomap). 
My question is for turtlebot3, we don't have turtlebot_bringup 3dsensor.launch file. How shall I turn on all the sensor processing modules as the default? Secondly, for turtlebot2, we need to set the environmental variables for the Kinect sensors are as follows:
export KINECT_DRIVER=freenectexport 
TURTLEBOT_3D_SENSOR=kinect
Do I need to do this kind of seting up for Turtlebot3? 
Thank you so much for your help. After I figure it out completely, I would right a tutorial to share my experience to reduce your workload on explaining 3D Visualization with turtlebot3. Thank you again.

The following is part of the tutorial I am follwoing: 
Visual data using rviz with Turtlebot

To visualize the 3D sensor data from the TurtleBot using rviz, begin by launching the TurtleBot minimal launch software. Next, a second terminal window should be opened to start the launch software for the 3D sensor:
Terminal Window 1: Minimal launch of TurtleBot:
$ ssh <username>@<TurtleBot's IP Address>
$ roslaunch turtlebot_bringup minimal.launch
Terminal Window 2: Launch 3D sensor software:
$ ssh <username>@<TurtleBot's IP Address>
$ roslaunch turtlebot_bringup 3dsensor.launch
The 3dsensor.launch file within the turtlebot_bringup package configures itself based on the TURTLEBOT_3D_SENSOR environment variable set by the user. Using this variable, it includes a custom Kinect or ASUS Xtion PRO or RealSense R200 launch.xml file that contains all of the unique camera and processing parameters set for that particular 3D sensor. The 3dsensor.launch file turns on all the sensor processing modules as the default. These modules include the following:
rgb_processingir_processing
depth_processing
depth_registered_processing
disparity_processing
disparity_registered_processing
scan_processing

It is typically not desirable to generate so much sensor data for an application. The 3dsensor.launch file allows users to set arguments to minimize the amount of sensor data generated. Typically, TurtleBot applications only turn on the sensor data needed in order to minimize the amount of processing performed. This is done by setting selected roslaunch arguments to false when particular sensor data is not needed.

Next, launch rviz to allow various forms of visualization data to be seen.
Terminal Window 3: View sensor data on rviz:
$ roslaunch turtlebot_rviz_launchers view_robot.launch

stevensu1838, 2018-07-25:
Hi sir,
I followed you instructions to use Kinect with Rivz for 3D vision on turtlebot3. 
You have to install freenect packages for using kinect.
$ sudo apt-get install ros-kinetic-freenect*
and run the camera.
$ roslaunch freenect_launch freenect.launch

I followed you advice above on how to use Kinect with Rviz for 3D vision and I've got point cloud image in Rivz. However the camera_link is always on the ground. How can I move the camare_link up? Coz I want to put the Kinect onto my mobile robot turtlebot3. The camera_link should be above the ground for sure. Thanks a lot for your help. Cheers.
Please check it out with the link below. 
[https://drive.google.com/file/d/1pjWWA7cQOWE-Kx8VVUVsX6sC9qukOHOC/view?usp=sharing](url)

Secondly, right now in Fixed Frame in Rviz I have only links about the camera suck as camera_rgb_frame, how can I get also the turtlebot3 links here? By doing `$ roslaunch turtlebot3_fake turtlebot3_fake.launch?` I mean I want to have both the links of the Kinect and turtlebot3. Thank you so much for your help. I am almost there. Thank you with no end.

Steven
Steven

stevensu1838, 2018-07-25:
[https://drive.google.com/file/d/1pjWWA7cQOWE-Kx8VVUVsX6sC9qukOHOC/view?usp=sharing](url)

stevensu1838, 2018-07-25:
https://drive.google.com/file/d/1pjWWA7cQOWE-Kx8VVUVsX6sC9qukOHOC/view?usp=sharing

stevensu1838, 2018-07-25:
Try this one please: [https://drive.google.com/file/d/1pjWWA7cQOWE-Kx8VVUVsX6sC9qukOHOC/view?usp=sharing](url)

routiful, 2018-07-26:
You can make the camera link to set urdf file in `turtlebot3_description` package.

https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_description/urdf/turtlebot3_burger.urdf.xacro

or you can make [static transform publisher ](http://wiki.ros.org/tf#static_transform_publisher) by tf packages

stevensu1838, 2018-07-26:
Hi Sir,
Thank you so much for your help. It is hard for me as a begginner to write my own camera link. So I copied Waffle's camera link part and set it Burger's urdf file in turtlebot3_description package. And it gives me camera link(but I use this link for Kinect) in Rviz for turtlebot3 as I expected. Also, I  could see the point could very well. However, when I adjust the Kinect to make it look down onto the floor, the point cloud image of the floor goes up into the air. I mean they move in the opposite way. Can you please tell me how to fix it?  Thank you so much for your support. 
Steven
[image]

routiful, 2018-07-27:
If someone want to look reply for above question, please following #273

routiful, 2018-08-02:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-02 by routiful