ROBOTIS-GIT/turtlebot3 #257: Why turtlebot3_navigation package could run without source file of move_base and acml?

Kanelankai, 2018-07-18:
In turtlebot3_navigation package, I found the launch files of acml and move_base.
These launch files can be executed according to emanual->navigation.
However, I could not find any source files about acml and move_base in turtlebot3 package.
I'm just curious about it.
Could you introduce some links or material introducing how you realized it?
Thank you very much!

routiful, 2018-07-18:
Hi :)

I attached link for you

https://github.com/ros-planning/navigation

Kanelankai, 2018-07-18:
Thank you for replying.
I also downloaded this package.
In this package, source files for amcl and move_base could be found, but in turtlebot3_navigation package there are only launch files and yaml files for them.
I found that without the navigation package, turtlebot3_navigation could also realize functions of amcl, move_base and so on.
So could you tell me how you realize the functions of amcl and move_base just using launch files and yaml files?
Thank you very much.

routiful, 2018-07-18:
The navigation packages are already installed. Simply, the launch file make execute multiple nodes more easily with some arguments. 

If you want to more detailed information about how launch file is working, please follow attached link below. After that, if you have any question or difficult understand, please ask me again.
 
[ROS WiKi](http://wiki.ros.org/roslaunch)

[ROS Robot Programming books](https://community.robotsource.org/t/download-the-ros-robot-programming-book-for-free/51)

[Free online course](https://community.robotsource.org/t/video-tutorials-covering-each-chapter-of-ros-robot-programming-handbook/1454)

Kanelankai, 2018-07-18:
Thank you very much!
I will confirm the links you gave.

routiful, 2018-07-18:
Thanks :) If you have any further issues, please feel free to get new ticket.

closed 2018-07-18 by routiful