ROBOTIS-GIT/turtlebot3 #259: OpenCR board no longer showing up on Serial after uploading Manipulator image

erikgroving, 2018-07-18:
ISSUE TEMPLATE ver. 0.1.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [X] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [X] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

    - Software version: [ROS Kinetic 1.12.13]
    - Firmware version: [1.2.0]
 

6. Please, describe detailedly what difficulty you are in 
Hi, I am having a similar issue to #240. I commented out the same line of code because it wasn't compiling: `DEBUG_SERIAL.print("Bumper : "); DEBUG_SERIAL.println(sensors.getPushedBumper());
`
I also changed the joints to 5 and the grip joints to 1, and renamed the joints in the image since I have a custom arm. After uploading the arduino image, the OpenCR board no longer shows up on Linux and Windows outputs a USB device not recognized error message.

I changed the code to the recommended `checkPushBumper `function but the board does not show up anymore. It showed up fine before I uploaded the image.

erikgroving, 2018-07-18:
Slight update, was able to get it to show up on linux by making the OpenCR board enter DFU_Util mode:
`Bus 005 Device 030: ID 0483:5740 STMicroelectronics STM32F407`
I subsequently tried to put the core example on the board but received the following output:
```
Sketch uses 180072 bytes (22%) of program storage space. Maximum is 786432 bytes.
Global variables use 62116 bytes of dynamic memory.
opencr_ld ver 1.0.2
opencr_ld_main 
An error occurred while uploading the sketch
>>
file name : /tmp/arduino_build_206333/turtlebot3_core.ino.bin 
file size : 177 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
cmd_read_board_name fail : 0xF020

```

When I try to do Burn Bootloader, the output is as follows:
```

dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
Error while burning bootloader.

```

erikgroving, 2018-07-18:
Well, I seem to have fixed this issue. Not quite sure what fixed it exactly, but some combination of entering DFU mode and Firmware recovery mode seemed to do the trick.

closed 2018-07-18 by erikgroving

routiful, 2018-07-19:
If you have any question about it or repeat again, please re-open this issue.

timdori, 2018-08-15:
My problem is exactly the same but i am not able to solve it. I tried different variations of firmware recovery mode and entering the dfu mode. The command lsusb is not showing the STM device. When I try to enter DFU mode lsusb doesnt show a bus in dfu mode. Please can anyone help me?

routiful, 2018-08-16:
You are supposed to push button before enter dfu mode.
Please follow attached link.

http://emanual.robotis.com/docs/en/parts/controller/opencr10/#bootloader

timdori, 2018-08-16:
Thank you for the fast reply. I entered the DFU mode and alle led went off only the first one ist on. I think it was right but its still not listet with lsusb. Any other ideas?

OpusK, 2018-08-17:
Hi, @timdori.

Can not proceed in [E-manual way](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#update-bootloaderlinux)(including dfu-util installation)?

timdori, 2018-08-17:
Hey, i tried to. But it is still not shown. When i connect to win7 pc the devicemanager shows unknown device and the hardware id is also unknown. It has to be a hardware problem or does anyone has another idea?

erikgroving, 2018-08-17:
Firmware-recovery mode is what has worked for me. 

> Hold down the PUSH SW2 button.
> Press the Reset button.
> Release the Reset button.
> Release the PUSH SW2 button.

Good luck.

OpusK, 2018-08-18:
@timdori 

Is your operating system Windows?

If so, Have you tried the method in [this link](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#install-windows-driver)?

timdori, 2018-09-10:
Hi thank you very much for your help. I bought another OpenCr board and everything was fine. So i think it has to be an hardware issue.