ROBOTIS-GIT/turtlebot3 #260: Tutlebot3 burger bring up error

babe1031, 2018-07-18:
Hi Team,

I put on a Arducam camera(OV5647 for Raspberry Pi) on my burger. Then I try to bring up and it throws errors as below. And I have also a second ROS working space for UR5 robot apart from my cakin_ws  working space where I put Turtlebot packages, not sure which is the root cause of the errors. Thank you so much for your support.
```
user@mech1349:~/catkin_ws/src$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/user/.ros/log/196c7bd8-8a57-11e8-8168-4c72b9268f33/roslaunch-mech1349-4691.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://132.181.61.226:38348/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://132.181.61.226:11311

process[turtlebot3_core-1]: started with pid [4708]
ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/user/catkin_ws/src
ROS path [2]=/home/user/ros_industrial_ws/src
ROS path [3]=/opt/ros/kinetic/share
process[turtlebot3_diagnostics-3]: started with pid [4709]
[INFO] [1531897020.069137]: ROS Serial Python Node
[INFO] [1531897020.076820]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1531897020.078727]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
[turtlebot3_core-1] process has finished cleanly
log file: /home/user/.ros/log/196c7bd8-8a57-11e8-8168-4c72b9268f33/turtlebot3_core-1*.log
```

robotpilot, 2018-07-18:
Hi @babe1031,

1. There is no package called `hls_lfcd_lds_driver`. You need to install it.
```
ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver
```
Please refer to the e-manual below.
http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#3-install-dependent-packages-on-turtlebot-pc

2. I think there are two things I can guess from this error. 1) The OpenCR board is not connected to the computer running this launch file. 2) The OpenCR board may be connected to other port, not '/ttyACM0'.
```
[ERROR] [1531897020.078727]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
```

babe1031, 2018-07-19:
Hi Team,
The problem is fixed and it was because I didn't do $ ssh <username>@<IP address of TurtleBot> to connect the robot. I don't see those errors again. However, now I get errors for the camera and as i said before I put on a Arducam camera(OV5647 for Raspberry Pi) on my burger. Now after I run roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch, I get the following errors:
```
... logging to /home/turtlebot3/.ros/log/8f4248a0-8ae4-11e8-a0fa-4c72b9268f33/roslaunch-turtlebot3-1486.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.32.51.156:44187/

SUMMARY
========

PARAMETERS
 * /raspicam_node/camera_frame_id: camera
 * /raspicam_node/camera_info_url: package://turtleb...
 * /raspicam_node/framerate: 15
 * /raspicam_node/height: 480
 * /raspicam_node/width: 640
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    raspicam_node (raspicam_node/raspicam_node)

ROS_MASTER_URI=http://132.181.61.226:11311

ERROR: cannot launch node of type [raspicam_node/raspicam_node]: raspicam_node
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/turtlebot3/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
turtlebot3@turtlebot3:~$ 
```

Here, I show you all the lines that I tap in:
```
terminal1: roscore
terminal2: ssh <username>@<IP address of TurtleBot>
                   roslaunch turtlebot3_bringup turtlebot3_robot.launch
terminal3: export TURTLEBOT3_MODEL=burger
                   roslaunch turtlebot3_bringup turtlebot3_model.launch
All is good so far. Then I get errors when i run in terminal4 ssh <username>@<IP address of TurtleBot> and roslaunch turtlebot3_bringup turtlebot3_rpicamera.launch
Thank you so much for your support.
```

robotpilot, 2018-07-19:
- The message below is that the **raspicam_node** package is not installed.
```
ERROR: cannot launch node of type [raspicam_node/raspicam_node]: raspicam_node
```
- If you want to fix this error, you need to install **raspicam_node** on your SBC of TurtleBot3. Please refer to the description below for installation of the package.

- http://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/#installation

- Note that the above raspicam_node package works well with Raspberry Pi Camera. I don't know if Arducam you mentioned will work.

babe1031, 2018-07-21:
Hi Sir,
The problem is fixed after I follow you instructions. And I could see the image in image view. However, I could not receive any images in the Image topic window in Rviz, can you please take a the linked the screenshots of the rqt_graph and errors in Rivz? I hope to see the image in Rviz. Can you please help me out? Thanks a lot. [Cheers](https://drive.google.com/drive/folders/1Zlt9YOEjMa14U0zXZZScLigAfKrv2wVI?usp=sharing)

babe1031, 2018-07-21:
Sir, 
Please find the linked the pictures here[](https://drive.google.com/drive/folders/1Zlt9YOEjMa14U0zXZZScLigAfKrv2wVI?usp=sharingurl)

babe1031, 2018-07-21:
Hi Sir,
Please find attached the link. Cheers
[https://drive.google.com/drive/folders/1Zlt9YOEjMa14U0zXZZScLigAfKrv2wVI?usp=sharing](url)

stevensu1838, 2018-07-21:
Hi Sir, please click the highlighyed cheers in the above comment to check the screenshots  Thanks a lot

robotpilot, 2018-07-22:
The topic name is blank. Please enter the name of the image topic you are using.
[image]

routiful, 2018-08-03:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-03 by routiful