ROBOTIS-GIT/turtlebot3 #262: TB3 burger is going opposite direction when I move forward by andriod app (the picture say it all)

angrysword, 2018-07-19:
ISSUE TEMPLATE ver. 0.1.0

1. Which TurtleBot3 you have?

    - [x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ X] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

3. Write down software version and firmware version

    - Software version: [1.12.12]
    - Firmware version: [x.x.x]
 
4. Write down the commands you used in order

    [Remote PC] roscore
    [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
 Hi Team

 I use android teleop app. and when I ask turtlebot move foreward. however turtle bot going backward. 

Please refer the picture. When I click blue circle in the phone, it suppose move foreward, however it move backward.  This problem will leads the slam  follower app can not work correctly. 

Thanks guys

[image]

kijongGil, 2018-07-19:
Hi @angrysword :)
Could you check your DXL's ID? Letf DXL is 1, right DLS is 2.
Thanks,
Gilbert.

angrysword, 2018-07-19:
Thanks Gilbert, you are right, you save me a day and a day ahead and .... big thanks.  Works now

closed 2018-07-19 by angrysword

kijongGil, 2018-07-20:
Thanks, If you have any further questions, please feel free to open new issue page.