ROBOTIS-GIT/turtlebot3 #263: /cmd_vel topic doesn't move the dynamixels

TheBard2273, 2018-07-20:
ISSUE TEMPLATE ver. 0.1.0

1. Which TurtleBot3 you have?

Burger

2. Which SBC(Single Board Computer) is working on TurtleBot3?

Raspberry Pi 3

3. Which OS you installed in SBC?

Ubuntu MATE 16.04

4. Which OS you installed in Remote PC?

Ubuntu 16.04 LTS (Xenial Xerus)

5. Write down the commands you used in order

    [Remote PC] roscore
    [SBC] roslaunch turtlebot3_bringup turtlebot3_robot.launch
    [Remote PC] roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
 
6. Copy and Paste your error message on terminal

   None
  
7. Please, describe detailedly what difficulty you are in 

    -Hi,

I have a problem with turtlebot3 movement and I was hoping I can get some help from you.
I am building a turtlebot3 but instead of making it with the black original frames that are portrayed in the website, I'm using my own patterned plates and brackets as the structure, so the only thing different about it is the structure. All the other components like dynamixels and boards and Lidar sensor are the same as in the ROBOTIS website.
I managed to bring the robot up until the end of the "Bring-Up" part on the e-manual, but when I want to move the robot using the "teleop" I don't get anything.
I have my turtlebot3 and remote PC connected through wifi, I can see that the "/cmd_vel" topic is being published in when I get the echo of the topic in the turtlebot PC, I see two of the red lights on the OpenCR board, which are indicating the number of users, get lit, but I have no movement in the robot because the dynamixels are staying stationary. I even tried publishing into the /cmd_vel topic manually in the turtlebot PC but that didn't work either.

Let me emphasize that the only difference between the original turtlebot3-burger and my turtlebot3 is the model of the structure and the related matters like URDF models and etc.
 
I will really appreciate the smallest help.
Thank you.

kijongGil, 2018-07-20:
Hi TheBard2273 :)
Did you check your Remote PC and turtlebot3's IP? It will be help you. 
https://github.com/ROBOTIS-GIT/turtlebot3/issues/229

If you can't solve your problem, please tell me again.
Thanks,
Gilbert

TheBard2273, 2018-07-20:
Yes, I checked the IPs and I've written them down correctly in ~/.bashrc. As I said the remote PC will publish in the /cmd_vel topic on the turtlebot without any problems, on the other hand I can see the "rqt" on the remote PC and every topic is being published in. Even the "imu" topic changes when I move the structure of the robot by hand, so I'm getting this idea that everything is working as they should be but the dynamixels are not working at all.
Let it be known that I have not altered any of the fundamental codes or anything. The only changes I made in the codes that I had were in the URDF-related codes so that I can show my own model in the rviz.

TheBard2273, 2018-07-20:
I must also add that I checked the dynamixels with Dynamixel Workbench using "dxl_to_usb" sketch on the OpenCR board and used velocity controller to move them, and they worked just fine.

TheBard2273, 2018-07-20:
I just tried to move the motors with the exact models of a turtlebot3 burger (I mean the stl files and URDF and everything) and it still doesn't work. The model in rviz doesn't move either even when I change the fixed frame to "odom"

kijongGil, 2018-07-20:
Please try again and check /cmd_vel publisher and subscriber when you publish this topic.

`$ rostopic info /cmd_vel`

Tell me what you see.

TheBard2273, 2018-07-20:
I ran that code multiple times after launching the proper launch files and it always shows "turtlebot3_teleop" as the publisher and "turtlebot3_core" as the subscriber

TheBard2273, 2018-07-20:
Another piece of information that might give you a better insight for helping me further is that after all of the OpenCR setup that I have gone through, by pushing and holding SW1 or SW2 on the OpenCR board, robot did not move at all:
http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#basic-operation

routiful, 2018-07-23:
Hi :)

Have you try to set ID for Dynamixels? I assume that Dynamixels have a problem to initialize.
Please follow below instruction and try test motor agian

http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3

TheBard2273, 2018-07-23:
It's funny that right after I did the exact procedure in the link and got good results I got this email !! ;))
But thank you so much for the help. Hope this solution can help more people in the future.

routiful, 2018-07-23:
Great! if you have any further question, please feel free to open issue :)

closed 2018-07-23 by routiful