ROBOTIS-GIT/turtlebot3 #268: Installation instructions

pitosalas, 2018-07-24:
In the book you state that this will do a complete installation of kinetic, rod, and tb3 components:

```
$ wget https://raw.githubusercontent.com/ROBOTIS-GIT/robotis_tools/master/install_ros_kinetic.sh
$ chmod 755 ./install_ros_kinetic.sh
$ bash ./install_ros_kinetic.sh
```

Does that procedure take care of all the steps that are in the TUrtlebot3 Wiki at this url: http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#pc-setup

Because a single comprehensive master install script would be really nice!

robotpilot, 2018-07-24:
Hi @pitosalas
Do you think it's better to have everything installed at once? 
I would like to have some flexibility. :)

pitosalas, 2018-07-24:
I ask because the install of ROS is so delicate that when I see two different procedures in totally different places, I wonder whether they are different, one is obsolete, or one is preferred.

robotpilot, 2018-07-24:
The two installation methods mentioned above are the same. The install_ros_kinetic.sh file installs only the ROS.

routiful, 2018-08-02:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-02 by routiful