ROBOTIS-GIT/turtlebot3 #269: Navigation issue

zijian-xu, 2018-07-24:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ✓ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ✓  ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

3. Which OS you installed in SBC?

    - [ ✓  ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

4. Which OS you installed in Remote PC?

    - [ ✓  ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order
[Remote PC]
     - roscore

[ TB3 ]
     - roslaunch turtlebot3_bringup turtlebot3_robot.launch

[Remote PC]
     - roslaunch turtlebot3_slam turtlebot3_slam.launch
     - rosrun map_server map_saver -f ~/map
     - then I just killed the slam task

[Remote PC]
     - roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml

7. Copy and Paste your error message on terminal
    - [WARN] Failed to get param: timeout expired 
          -- I get this error when  launching roslaunch turtlebot3_bringup turtlebot3_robot.launch

    - "could not get robot pose, cancelling reconfiguration"
    - "Costmap2DROS transform timeout. Current time: 153253596.3332, global_pose_stamp: 
        1532353713.8062, tolerance: 0.5000"
    - Extrapolation Error looking up robot pose: Lookup would require extrapolation into the past
       Requested time 1532416935.14169903 but the earliest data is at time 1532416953.542184321,
       when looling up transformation from frame[base_footprint] to frame[map].
        -- I get these warnings/errors when running navigation task.
  
8. Please, describe detailedly what difficulty you are in 

    - When I do the slam task, the orientation of the robot shown in rviz is the same as the robot which is placed in our test environment. The map the LIDAR scans out was also accurate(at least from what I observed in RVIZ).
    - I did the navigation task then. First I did the 2d Pose estimation, the orientation tb3 shown in rviz was opposite direction.(in real world, tb3 is facing to the right-hand side, when I use 2d pose estimation in rviz and pointed the direction to the right-hand side, the robot faced to the opposite direction). 
   - And then I just set the navigation goal like moving to the right hand side, the robot instead moved to the left hand side.

-- Photo of RVIZ after clicking 2d Pose Estimation
[image]

-- Photo of robots in real world
[image]

zijian-xu, 2018-07-24:
Thanks in advance!

zijian-xu, 2018-07-24:
Problem solved, I just exchanged the two motors and the navigation worked. Thank you

robotpilot, 2018-07-24:
I hope everything is going great. I will close this issue. If you have any question or suggest, please ask me any time. Thanks!  :)

closed 2018-07-24 by robotpilot