ROBOTIS-GIT/turtlebot3 #271: rostopic shows /odom publisher and subscriber, but topic monitor shows "unknown"

pitosalas, 2018-07-24:
Trying to get a new robot up and running. When

```
rostopic info /odom
```
I see both a publisher and subscriber, but, rqt topic monitor shows /odom (and all other topics) to be   with unknown values.

If this matters, we don't have the Lidar attached and during bringup there is an error but I was assuming this was not the cause. See the output of bringup:

```
roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/robot/.ros/log/b9caeb96-8f70-11e8-bce4-4437e6b74bd1/roslaunch-donatello-2966.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:40822/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix:
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://roscore1.cs.brandeis.edu:11311

process[turtlebot3_core-1]: started with pid [2976]
process[turtlebot3_lds-2]: started with pid [2977]
process[turtlebot3_diagnostics-3]: started with pid [2978]
[ERROR] [1532458956.235254716]: An exception was thrown: open: No such file or directory
[turtlebot3_lds-2] process has died [pid 2977, exit code 255, cmd /home/robot/catkin_ws/devel/lib/hls_lfcd_lds_driver/hlds_laser_publisher __name:=turtlebot3_lds __log:=/home/robot/.ros/log/b9caeb96-8f70-11e8-bce4-4437e6b74bd1/turtlebot3_lds-2.log].
log file: /home/robot/.ros/log/b9caeb96-8f70-11e8-bce4-4437e6b74bd1/turtlebot3_lds-2*.log
[INFO] [1532458957.120586]: ROS Serial Python Node
[INFO] [1532458957.164305]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1532458959.304822]: Note: publish buffer size is 1024 bytes
[INFO] [1532458959.306076]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1532458959.322023]: Setup publisher on version_info [turtlebot3_msgs/VersionInfo]
[INFO] [1532458959.375162]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1532458959.388988]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1532458959.411120]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1532458959.425082]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1532458959.438586]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1532458959.452736]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1532458961.739648]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1532458961.761586]: Note: subscribe buffer size is 1024 bytes
[INFO] [1532458961.762970]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1532458961.786337]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1532458961.807380]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1532458961.830382]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1532458962.368968]: --------------------------
[INFO] [1532458962.371672]: Connected to OpenCR board!
[INFO] [1532458962.374400]: This core(v1.2.0) is compatible with TB3 Burger
[INFO] [1532458962.377094]: --------------------------
[INFO] [1532458962.379751]: Start Calibration of Gyro
[INFO] [1532458964.925763]: Calibration End

```

robotpilot, 2018-07-24:
Hi @pitosalas 
> I see both a publisher and subscriber, but, rqt topic monitor shows /odom (and all other topics) to be with unknown values.

There is a check box for each topic in the 'rqt topic monitor'. You can see the topic value by clicking on the checkbox. Did you do that?

http://emanual.robotis.com/docs/en/platform/turtlebot3/topic_monitor/#topic-monitor

pitosalas, 2018-07-25:
Thanks @Pyo:

Yes I did set the check mark. In several places to try, battery, tf, odom. 

Also I verified that when I shutdown the Turtlebot all those topics disappeared off the list.

But none of them was updating with any data.

Also when I used the teleop utility the robot didn’t move even though the cmd_vels were being published. 

Not sure how to troubleshoot further… Any thoughs?
Pito Salas
Brandeis Computer Science
Volen 134

kijongGil, 2018-07-26:
Hi @pitosalas :)
It seems that turtlebot3 and remote PC are not connected.
Could you check again?
https://github.com/ROBOTIS-GIT/turtlebot3/issues/244

pitosalas, 2018-07-26:
I will check again tomorrow, but here's what I believe to be true:

- TB3 is on Wifi
- RemotePC is on Wifi
- Roscore is on wired network in the same subnet
- They can all three ping each other
- When robot is up, its topics e.g. /odom are displayed with rostopic list
- When robot is shutdown they disappear (from rostopic list as well as from rqt topic monitor
- rostopic list on robot works (so it can see Roscore)

Would that be consistent with them not being connected?

pitosalas, 2018-07-27:
@pyo well the problem was that I had used "localhost" for ip_hostname instead of a real IP address. Making that change solved it.

closed 2018-07-27 by pitosalas