ROBOTIS-GIT/turtlebot3 #273: How to use Kinect sensor URDF code from Turtlebot2 in my Turtlbot3 URDF

stevensu1838, 2018-07-26:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ ] Burger

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - NUC

3. Which OS you installed in SBC?

    - Ubuntu  16.04 
  

4. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)

5. Please, describe detailedly what difficulty you are in 

    - 
Dear Pyo
I am trying to put Kinect on my turtlebot3 for 3D vision and right now I need to make the camera link to set urdf file in turtlebot3_description package. I plan to copy the Kinect sensor URDF code from Trutlebot2 to Turtlebot3 URDF file. However, the URDF files for the two robots are organised into ways and it is very hard to figure out how to do the copy. I know you’re an expert for both turtlebot2 and tutlebot3. Can you please show mE some advice on how to use the Kinect sensor code from Turtlebot2 in my turtlbot3 URDF? Thanks a lot.

Best regards,

Steven

routiful, 2018-07-27:
I give an example how to set `camera joint` in URDF.

https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_description/urdf/turtlebot3_burger_for_autorace.urdf.xacro#L165

The burger used in turtlebot3_autorace has camera to detect lines.
So this URDF including `camera joint` is different from original URDF for burger.

Before you hard copy, I recommend to read some tutorials 

http://wiki.ros.org/urdf/Tutorials/Create%20your%20own%20urdf%20file

stevensu1838, 2018-08-12:
Hi,
May I please ask why do I have to include relevant gazebo.xacro files into the urdf.xacro files in the beginning? Do Gazebo and Rviz use this same urdf.xacro file? Or we only include the gazebo.xacro files for the use of robot visualization in Gazebo? Thanks.
Below is just an example of including a relevant gazebo.xacro file into the urdf.xacro file
<xacro:include filename="$(find turtlebot3_description)/urdf/turtlebot3_waffle.gazebo.xacro"/>