ROBOTIS-GIT/turtlebot3 #274: Teleop problem

space03, 2018-07-26:
ISSUE TEMPLATE ver. 0.2.0

1. Which TurtleBot3 you have?

    - [ v] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [v ] etc (PLEASE, WRITE DOWN YOUR SBC HERE) JetsonTX2 DeveloperKit

3. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Raspbian
    - [v] etc (PLEASE, WRITE DOWN YOUR OS) Ubuntu 16.04

4. Which OS you installed in Remote PC?

    - [v ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
 
6. Write down the commands you used in order

    - $ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch 

7. Copy and Paste your error message on terminal

    - HERE
  
8. Please, describe detailedly what difficulty you are in 

    - If I input 'w' key, turtlebot's left wheel goes to turn forward but right wheel goes to backward. 
'a' and 'd' key works well, but 'w' and 'x' key are weird. 

teleop terminal shows 
currently:	linear vel 0.01	 angular vel 0.0 
currently:	linear vel 0.02	 angular vel 0.0 
currently:	linear vel 0.03	 angular vel 0.0 
but the robot just turn to right side.

i'm not good at ros system and arduino. so please help me.

kijongGil, 2018-07-27:
Hi @space03 :)
Could you update OpenCR and turtlebot3's packages?
http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup

Thanks,
Gilbert.

space03, 2018-07-27:
Thank you for reply.
I make up this problem. It is a Dynamixel setting problem. 
from r+manager 2.0, it solved by setting right motor to id:2 and reverser mode.

Thank you.

kijongGil, 2018-07-27:
Great! if you have any further question, please feel free to open issue :)

closed 2018-07-27 by kijongGil