ROBOTIS-GIT/turtlebot3 #275: Turtlebot 3 Slam

pitosalas, 2018-07-27:
I'm following along the TB3 tutorial wiki and experimenting with Slam. It is working ok. I am seeing the edge conditions and saw various odd warnings which are concerning:

```
$ ros_wtf
ERROR The following nodes should be connected but aren't:
 * /move_base->/move_base (/move_base/global_costmap/footprint)
 * /move_base->/move_base (/move_base/local_costmap/footprint)
```

Also from `roslaunch turtlebot3 navigation turtlebot3_navigation.launch map file:=... I got:

```
-- 4172.2622, global_pose stamp: 1532714171.7426, tolerance: 0.5000
[ WARN] [1532714172.262382062]: Could not get robot pose, cancelling reconfiguration
[ WARN] [1532714173.270830690]: Costmap2DROS transform timeout. Current time: 1532714173.2707, global_pose stamp: 1532714172.7566, tolerance: 0.5000
[ WARN] [1532714173.362163917]: Could not get robot pose, cancelling reconfiguration
[ WARN] [1532714174.273314850]: Costmap2DROS transform timeout. Current time: 1532714174.2732, global_pose stamp: 1532714172.4053, tolerance: 0.5000
[ WARN] [1532714174.461652581]: Could not get robot pose, cancelling reconfiguration
```
and 
```
[ WARN] [1532714689.861442860]: Could not get robot pose, cancelling reconfiguration
[ WARN] [1532714690.862292205]: Costmap2DROS transform timeout. Current time: 1532714690.8622, global_pose stamp: 1532714689.3163, tolerance: 0.5000
[ WARN] [1532714690.862362027]: Could not get robot pose, cancelling reconfiguration
[ WARN] [1532714691.873289155]: Costmap2DROS transform timeout. Current time: 1532714691.8732, global_pose stamp: 1532714689.3163, tolerance: 0.5000
[ WARN] [1532714691.961505715]: Could not get robot pose, cancelling reconfiguration
[ WARN] [1532714692.873500408]: Costmap2DROS transform timeout. Current time: 1532714692.8733, global_pose stamp: 1532714689.3163, tolerance: 0.5000
[ WARN] [1532714692.962005510]: Could not get robot pose, cancelling reconfiguration
[ WARN] [1532714693.961530347]: Costmap2DROS transform timeout. Current time: 1532714693.9615, global_pose stamp: 1532714689.5175, tolerance: 0.5000
```

Can you tell me what those mean and how to fix them? Thanks!

routiful, 2018-08-02:
Hi :)

The problem can be occured because asynchronous time between SBC and remotePC
or bad condition of Wifi.

I recommended you are supposed to synchronize the time using ntpdate commands
http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

robotpilot, 2018-08-08:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

routiful, 2018-08-09:
Of course. If you want to auto execute it, you could make shell file and copy and paste it into `/etc/init.d` folder.

pitosalas, 2018-08-10:
But you could have given a little more detail. What do you mean by make a shell file? Shell file containing what? How do you mean "copy and paste it (the file) into /etc/init.d folder"?

closed 2018-08-10 by pitosalas