ROBOTIS-GIT/turtlebot3 #276: Turtlebot behavior in the real world

pitosalas, 2018-07-30:
Tell me if this is the wrong place to post this question.

I am working with my Turtlebot3 Burger, in a 4 x 10m room that is carpeted. 

When driving forward the robot does ok, but sometimes it seems to stick and stutter.

When driving backwards, it's pretty terrible. It buckles and bounces and moves very roughly.

What might be causing this?

Do you not qualify the TB3 for riding on carpeted surfaces? 

Thanks!

routiful, 2018-08-02:
Hi :)

We qualify that TB3 burger can ride on the carpet(short fur, <0.5cm) due to low base.
If you used carpet has long fur, it is possible to face that problem.

pitosalas, 2018-08-02:
Thanks. My carpet has very short fur, might there be another reason causing this problem?

Pito Salas
Brandeis Computer Science
Volen 134

routiful, 2018-08-03:
We tested on various carpets but we can't find any problem during riding.
But we can't qualify all carpets. Sorry for inconvenience.

pitosalas, 2018-08-04:
I am assuming that the carpet is not the problem, but what else could cause stuttering kind of motion? Could it be a time sync problem?

Pito Salas
Brandeis Computer Science
Volen 134

routiful, 2018-08-05:
It might be caused by time sync. Please follow attached link

http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

closed 2018-08-10 by pitosalas